retrospy/RetroSpy #45: RetroSpy not properly working with Raphnet GC to N64

TyMiles2012, 2019-02-11:
Hi!  I've been having an issue with RetroSpy working with my GC controller through this adapter.  It works just fine with the GameCube Controller via official connections such as GameCube, Wii and Switch via adapter, but if I use the Raphnet adapter, it doesn't work.  Here's what I've tested.

Spliced GC extension > RetroSpy > Raphnet > N64.  Firmware set to GC mode.

Most packets seem to be dropping as the inputs go through, but it seems to be doing so at what looks like a very low framerate.  Original NintendoSpy doesn't even pick it up and shows inputs locked, such as A+B pressed, or pressing A tilts C Stick, etc.  Setting to N64 mode doesn't work either.

GameCube Controller > Raphnet Adapter spliced > RetroSpy > N64.  Set to N64 mode.

This sort of works.  Inputs on Super Mario 64 register no problem, but it doesn't work with Ocarina of Time.  Jittery and inputs all over the place even with blink reduction.  Problem with this is that even if I worked, I would have to have it show N64 inputs.

I can confirm this is a Raphnet issue because I spliced my N64 controller directly and RetroSpy works without a hitch that way, plus GC works on Switch/Wii as mentioned above.

zoggins, 2019-02-12:
I just ordered one of the adapters and will take a look as soon as it gets here!  Thanks!

TyMiles2012, 2019-02-12:
Awesome!  Thank you for looking into it!

TyMiles2012, 2019-02-17:
Oh, I might as well show you footage as I had just gotten around to it.  This is the first scenario with the gamecube mode, in regards to the rate of packets dropping.  It just seems like inputs are being shown at a low framerate.  If there's anything you want me to try in the firmware while you wait for your adapter to arrive, let me know.  Currently using the one that came with RetroSpy.  Hasn't been modified.  Also captured my cursor overtop to show it's 60fps footage.

Here's the footage.
https://streamable.com/fg9hn

zoggins, 2019-02-22:
I have a good news bad news situation on this one.  I found the issue with the "low framerate" and figured out a fix.  The bad news is that the analog sticks and triggers are super jittery and I can't figure out why.  Its not bad cable reads, because its only the analog controls and not the digital buttons.  Obviously this jitter is not picked up by the N64 so I have to figure out where it is coming from.

zoggins, 2019-02-22:
should be fixed in 2.9.1.  Requires both a firmware update and viewer update.  Closing this, but let me know if you have anymore problems.

closed 2019-02-22 by zoggins