retrospy/RetroSpy #46: nintendo wii/nes classic/snes classic controller support

texztech, 2019-02-13:
I would like to know if we could add the nes/snes/wii classic controller to this project?

Thank you for your time.

zoggins, 2019-02-13:
Which controller?  What console?  Thanks.

zoggins, 2019-02-13:
If you mean for the Wii extension controllers.  The thought never occurred to me before to try.  I will do some research.  Thanks.

zoggins, 2019-02-13:
After some reading I THINK this is possible.  I have some parts coming and will let you know.

texztech, 2019-02-14:
Thanks if you want me to throw some coinage your way to help out let me know.

zoggins, 2019-02-27:
This work is almost done.  I am not super happy with how the encryption/decryption is handled some sometimes with changing games you have to unplug and replug the controller but once in a game and the controller/console handshake has been well established it works really well.  The nunchuck seems to be more prone to handshake issues than the classic controllers.

texztech, 2019-02-27:
Thank you for the work so far :P

zoggins, 2019-03-03:
Well NES/SNES Classic work awesome.  Wii is pretty good and Wii U is ok.  3rd party wiimotes are hot garbage and I am not sure I can make them work, but I have gotten the old non-Wii motion plus OEM wiimotes (with and without the wii motion plus addon) working as well as the newer ones with the built in Wii motion plus.  I would say its about 95%.  Usually anything weird can be fixed by disconnecting and reconnecting the controller.  Going to release soon.

closed 2019-05-21 by zoggins