retrospy/RetroSpy #58: Nintendo switch pro Support

rainzzz, 2019-03-09:
I would like to know if we could add the nintendo switch classic controller to this project?

Thank you for your time.

zoggins, 2019-03-09:
Unfortunately if there is no controller cable there is no way to spy on the controller.  Is there a switch controller with a cable?  I thought they were all wireless.

rainzzz, 2019-03-10:
> Unfortunately if there is no controller cable there is no way to spy on the controller. Is there a switch controller with a cable? I thought they were all wireless.

Yes, the switch handle can be connected not only by wireless but also by wire.
I really want it

rainzzz, 2019-03-10:
> Unfortunately if there is no controller cable there is no way to spy on the controller. Is there a switch controller with a cable? I thought they were all wireless.

I tried to use the android gc mode on the switch pro handle, but the button is jumping and not displaying properly.

zoggins, 2019-03-10:
I will look into it!  Thanks!

rainzzz, 2019-03-11:
> I will look into it! Thanks!

Thanks, thank you, I will always look forward to it. Thanks again.

rainzzz, 2019-05-13:
I hope I can do this myself. How can I do it at high speed? Or what knowledge should be understood.

zoggins, 2019-05-13:
Sorry, I haven't got a chance to look into it yet.  I moved and am still
unpacking.  The problem with anything USB is its really hard to spy on.  In
fact there is no open source sniffers available.  The only working options
at the moment cost 400 to 1200 dollars.  This is something I plan on
looking into this year, but I am not sure what hardware it is going to be.
If you want to start researching you need to look into sniffing USB
protocols.

zoggins, 2019-09-14:
This was relased with RetroSpy 3.0.

closed 2019-09-14 by zoggins