retrospy/RetroSpy #60: Saturn Mouse not detected as mouse by games in 3D Pad mode

Shentokk, 2019-03-11:
I tried this with Policenauts to get mouse to be read. It would read the buttons just fine, but I could not get it to recognize mouse movement in the game or on the input reader. I tried with other mouse supported games like Virtua Cop 2 and Planet Joker with the same results.

I am using v2.9.1 release with the 3D pad jumper. 3D pad functions as normal, but mouse appears to not work.

zoggins, 2019-03-11:
Will take a look

zoggins, 2019-03-11:
Assuming you were using the mouse skin?

zoggins, 2019-03-11:
The mouse didn't work in game either?  That is weird.  I am assuming it's the OEM mouse?

zoggins, 2019-03-11:
Does it work on the Saturn dashboard, no did inserted?

zoggins, 2019-03-12:
OEM Mouse.  Saturn Shuttle Mouse Skin. SATURN3D firmware.  Newest Firmware on the Uno, newest RetroSpy.  English translation of Policnauts, everything works fine for me.  Ummm, the newest firmware shouldn't matter, do you know what release the firmware is from or is it the newest?  Just curious, there shouldn't have been any changes that would affect the mouse.

Shentokk, 2019-03-12:
The firmware was from the release package. Mouse is OEM model 1 (grey).

zoggins, 2019-03-12:
And the buttons work, but not the movement!

Shentokk, 2019-03-13:
That's correct.

zoggins, 2019-05-21:
Still looking at this.  I cannot repro on my end.

Shentokk, 2019-05-22:
Is there any technical information I can pull to help narrow it down?

Shentokk, 2019-05-29:
Updated to v3.0 and mouse still doesn't seem to work. ABC & Start read properly, but the mouse portion doesn't function.

Shentokk, 2019-07-15:
So I just wanted to update that it was an issue with the cable. Got a new one and now it works fine.

zoggins, 2019-07-15:
Awesome!! Will close this issue.

closed 2019-07-15 by zoggins