retrospy/RetroSpy #61: Support for Gameboy Interface

nitrostemp, 2019-04-12:
would it be possible to add support for inputs thru a gba using GBI?

zoggins, 2019-04-13:
I am sure it is possible.  I just moved so all my equipment is in boxes, but as soon as I am settled I will take a look.

nitrostemp, 2019-04-15:
JRP told me this may help you in regards to the gba as a controller communication https://github.com/ExtremsCorner/gba-as-controller/blob/gc/controller/source/si.arm.c

zoggins, 2019-04-15:
Yeah, I got it working with the offical Gameboy Player software a while back, so I imagine it must be similar.

nitrostemp, 2019-06-10:
just checking up on this, did you make any progress?

zoggins, 2019-06-10:
Sorry for the delay.  I moved recently and am finally getting everything
unpacked.  Should be able to get to it this month. Again, sorry for the
delay life has been hectic.

zoggins, 2019-06-12:
Is there something that wasn't working?  I have use the stock GBA player disc and GBI and the Gameboy Advance works as expected.

nitrostemp, 2019-06-12:
im asking JRP, as he is the one that requested this; just to check what version of GBI are you using?

nitrostemp, 2019-06-12:
ok, ive been told that high fidelity might be different, attached is some context from JRP.
[image]

zoggins, 2019-06-12:
Not sure what version of GBI I have, it's older though, but with the 3.0
firmware and viewer gbihf works as expected. I will update GBI tonight to
the newest version and check that.

zoggins, 2019-06-12:
Just came home and checked.  Newest version of everything (UI, firmware and GBI) all seems to work great.

nitrostemp, 2019-06-13:
can you confirm it works with "gbihf.dol"

nitrostemp, 2019-06-13:
and im double checking, this is connecting a gba via the link cable to play..

zoggins, 2019-06-13:
Gbihf works this is via a gba hooked up via link cable to the second
joystick port

zoggins, 2019-06-13:
Ok that post was bad, it made no sense. :)
Ok, I test the newest version of gbi.dol and gbihr.dol (i did not test gbisr.dol because I forgot).
I was using an original Gameboy Advance hooked up to controller port 2 with a link cable.
Everything appeared to work correctly on the Everdrive menu and a GBA few games I started up.

closed 2020-08-16 by zoggins