retrospy/RetroSpy #62: SegaControllerSpy.cpp never reports controller disconnected status

RoberMC, 2019-05-17:
When Sega Genesis controller is disconnected, the function enters in an infinite while loop waiting for controller response that halts program flow indefinitely, it never reports controller disconnected status "SCS_CTL_ON = 0", and never skips the while loop.

zoggins, 2019-05-17:
Will take a look. Thanks!

zoggins, 2019-05-21:
Is this with a 3 or 6 button controller?  Just trying to figure out what while loop it is getting stuck on forever.  And this is the firmware that spins or the UI?  And plugging the controller back in doesn't resume functionality?  Just trying to narrow down the issue.  Thanks!

RoberMC, 2019-05-21:
I am talking about the firmware, the SegaControllerSpy.cpp file, and function SegaControllerSpy::getState(). 

It does not matter which controller, 3 or 6 buttons. The library uses "do whiles" to wait for different statuses of the controller to do its port maskings and stuff. 

If there is no controller at the start of the program, the "do" part of the first "do while" becomes an infinite loop as it seems the condition in the first "while" is always met. if you start the program with a controller plugged and then suddenly you unplug it, any "do while" can be the infinite loop, whichever was running when you pull off the cable. If you connect again the controller the program resumes fine.

You can see this happening putting Serial.prints in each "do" part of every "do while", and see how one of them repeats infinitely when the controller is disconnected not letting the program flow to continue.

I fixed it myself enabling interrupts and putting a timeout in the "do whiles", so it skips if more than 5 ms have passed since it started, but that makes the controller reading incorrect frequently, i wish i knew more about direct port manipulation :(

Thanks to you for this awesome piece of software, this is like the bible of classic controller reading. if i can help with something or need more info or testing, just let me know.

zoggins, 2019-05-21:
What is the result in viewer is this case?  Does it show incorrect stuff when it's infinite looping?  The while loops are intentional, but the assumption is if you plug back in the controller it's all good.  I just want to understand your use case and make it better. I am always looking for testers! 😁 What consoles do you have?

RoberMC, 2019-05-21:
Viewer works fine, this "problem" does not cause any "malfunction" as viewer just waits for serial information from the firmware, so it just sits silently waiting, or if you disconnect the gamepad while pressing a button, viewer shows the button is being pressed forever while the controller is disconnected, as the firmware never reports controller disconnected status as it is infinitely looping ;)

You are right, telling you my use case should be helpful. I am trying to use a single Arduino Mega for reading multiple controllers. Not at once, i wired a button to switch between available controllers; Genesis, Snes, and PlayStation... for now. So the problem came in as soon as i tried to use the SegaControllerSpy.cpp library in my project, as when the sega controller is not connected it infinite loops and interrupts the whole program flow of the whole sketch. The push button i use to select the controller stops working, so i cannot switch to another controller, color leds stop working.... well everything stops working till i plug a genesis controller.

I hope this helps you fixing the issue. Thank you very much.

zoggins, 2019-05-21:
That absolutely helps, let me think about the code a bit!

zoggins, 2019-05-22:
Actually think this is a problem for most controller types in the firmware, as they wall use while loops to wait for signals.  It just happens that PS and SNES have overlapping pins, so they are capable of unblocking each other in most cases.  The overall problem is MODE_DETECT was always assumed to require a board restart.  Let think about this some more.  One possibility is to use pin change interrupts instead of while loops, but the timing is pretty tight on genesis controllers, so that might be too much overhead.  I am definitely thinking about this and will come up with something.

RoberMC, 2019-05-23:
Apparently, this guy managed to get it working, but i tried to implement his code into my sketch and could not make it to work, also i do not undestand anything about his pad reading code.

https://github.com/SukkoPera/MegaDrivePlusPlus

closed 2021-12-01 by zoggins