retrospy/RetroSpy #66: Flicker and shadow input on nano

krusherpt, 2019-05-25:
So I'm having problems with my arduino nano clone and retrospy.
I already rechecked my soldering and wiring but still having flickering and shadow input, using the old nintendospy firmware I have less flicker or shadow input, I'm using an extension for the wiring.
Also it freezes the inputs every 1-2 seconds.
Can anyone help me please?

zoggins, 2019-05-25:
What version of the firmware are you using? You are just straight using
MODE_NES?  I am here to help, but on vacation with limited internet until
Tuesday.

krusherpt, 2019-05-25:
Im using MODE_SNES and thanks for the help.

krusherpt, 2019-05-25:
Also I edited the original post so, the freeze was from the game, nothing to do with the retrospy (when I die theres no input registering)

zoggins, 2019-05-25:
I have never used a nano, are you using the unmodified firmware on the
nano? Usually flickering/shadow inputs are due to interrupts not being
disabled when reading the controller, but they are off to SNES.

krusherpt, 2019-05-25:
yes  im using it unmodified, since the pinout and specs are the "same", I'm no expert on arduinos but the only thing that improved the situation was the 2014 nintendospy firmware, but still had flickers and shadow input.

zoggins, 2019-05-25:
Also exactly what board are you using, I can pick one up to see if I can
reproduce it

krusherpt, 2019-05-25:
Nano V3.0 Mini USB ATmega328 5V 16MHz Micro-Controller CH340G Driver For Arduino

zoggins, 2019-05-28:
Board ordered, will figure it out

zoggins, 2019-06-12:
I got the board and I cannot repro.  It seems to work great for me.  I used the ATMega382P (Old bootloader) to upload and it just works.  Is there a particular game you were seeing it, more than others?

edru, 2019-07-30:
Ive made numerous nano devices at this point and have seen no issues myself.

zoggins, 2020-05-16:
One case where i see this is when Arduino GND isn't connected to the controller GND, but I have no idea if that is the case here.  There is also another bug oN SNES about flickering when CPU is under heavy usage.

closed 2020-08-16 by zoggins