retrospy/RetroSpy #67: Wii controller viewer support for 8BitDo GBros. Adapter

TomBobBlender, 2019-06-03:
I would like to request a controller viewer for the GBros. adapters Wii port. I can confirm with you that the GameCube side of the GBros. adapter works perfectly using the Arduino Uno viewer software just fine. However plugging the viewer cable in with the first party Wii controllers displays all buttons pressed and the control sticks moving erratically. Thank you for all that you do!

zoggins, 2019-06-03:
What board are you using?  What version of RetroSpy and do you have a link
to the adapter?  Wii support first shipped in 3.0 and only is supported on
the Teensy board > 3.5.  I can get an adapter and test it for you.

TomBobBlender, 2019-06-03:
It is the teensy 3.6 using the newest 3.0 firmware you provided. Here is the link to the adapter: https://www.amazon.com/gp/product/B07JLF9PVG

zoggins, 2019-06-03:
Got one coming tomorrow, I will figure it out.  Did any Wii controller work
for you, or have you tried like a nunchuk or classic controller?  I am just
curious

TomBobBlender, 2019-06-04:
I tried a lot of different setups for you, and the NES, SNES, Classic, Classic Pro and the Nunchuck all performed very well and never lost connection. It even works with the Raphnet Wii to GCN/N64 controller converter as well ( https://www.raphnet-tech.com/products/gcn64_to_wiimote/index.php ). I can also confirm that it also works with the 3DS controller adapter mod that's out there.

zoggins, 2019-06-04:
That is a thorough testing!  I appreciate that.  I have not heard any
results in the wild of the Wii support, so good to hear it works somewhere
other than my machines. :)  Adapter came today, will get a chance this week
to look at it.

zoggins, 2019-09-15:
Sorry this took, so long, but a release that supports this will be out soon.

TomBobBlender, 2019-09-15:
Thanks for all of your hard work, @zoggins ! I really look forward to the future release!

closed 2019-10-03 by zoggins