retrospy/RetroSpy #68: [Feature request] Ability to use any skin with any controller type

codemann8, 2019-06-05:
I'm playing games on different emulators, but I'm using a generic PC controller.  It'd be nice to have the PlayStation controller skin appear when I'm playing on ePSXe and be able to switch the skin to a SNES controller when I switch to Snes9x emulator games.

zoggins, 2019-06-05:
I will think about how to do this, its a good idea though.  The hard part
is if people use different controller button mappings, but I will think
about it.

edru, 2019-07-30:
I'd like this too - or the ability to at least define multiple controller types in one skin so we can save on space.

One idea i had was that you can define all the types of buttons in one skin, either by multiple entires, or comma separated values for the button name itself. Based upon system you select it will only look at the values that are needed, ignoring any additional values that may be there. (ie, NES controller can be defined for Wii as well, so you could define the disconnect icon, but when using generic NES, it could ignore the defined values in the skin)