retrospy/RetroSpy #75: Previous Window resize request...

ZanasoBayncuh, 2019-07-07:
Was for the controller preview because it must remain up to be usable and it gobbles up 60% of the monitor. A great alternate though would be if it could produce something the Browser Plugin on OBS could use... so a visible window wouldn't be needed.

zoggins, 2019-07-07:
I will look at what that would require.  Thanks!

FerretBomb, 2019-11-25:
Just as a note, so long as you leave even a tiny part of the actual window on-screen (not the titlebar, the actual viewer window), OBS can window capture the entire thing. I usually leave it all the way bottom-left, mostly offscreen, with only the top-right 25px visible. If the window viewport is entirely offscreen the OS 'intelligently' stops rendering it. Just a workaround in the interim.

ZanasoBayncuh, 2019-11-27:
> Just as a note, so long as you leave even a tiny part of the actual window on-screen (not the titlebar, the actual viewer window), OBS can window capture the entire thing. I usually leave it all the [...]

I imagine that's a Windows 10 thing or some such. Because for me this is utterly false.

https://puu.sh/EJbkA/aa4b42cee9.mkv

FerretBomb, 2019-11-27:
It works on Windows 7 as well if you have Aero enabled (and are using an Aero theme), as most people should.

(The misinformation that disabling Aero improves performance persists from the early days of Vista; it's long since been patched to render that 'common knowledge' wrong on all but the lowest-end garbage of email/facebook-tier machines).