retrospy/RetroSpy #78: [Organization Request] Standardize system names for skins

edru, 2019-07-21:
I notice some skins call the system by a name, NES, the company, SEGA, and even company/name.

Request to unify this across the board to either abbreviation (NES GEN GCN), or Company + Name.

zoggins, 2019-07-21:
Are you using the newest version?  I did a cleanup and attempted to standardize the names in the newest release.

edru, 2019-07-21:
i mean more in the skin design itself.

for example:
<skin name="Default" author="40wattrange" type="nes">
<skin name="Default" author="40wattRange" type="sega">

And I swore there was one that included the entire company and system like "Sega Dreamcast" - thought it wasnt dreamcast.

This is literally a small thing - i mainly discovered it since genesis was labeled sega, and not genesis, so was getting confused. This can basically just be ignored.

zoggins, 2019-09-16:
The outlier was Genesis which was just labeled Sega.  I have changed it.

closed 2019-10-03 by zoggins