retrospy/RetroSpy #80: Static Name for Viewer Window

edru, 2019-07-21:
OBS captures windows via the name of the window. RetroSpy's viewer window title changes based on the skin name. This causes users to have to change the window OBS captures each time they change to a different console.

To combat this it would be nice if the viewer window had a static name that never changed. As the initial window of RetroSpy is titled RetroSpy, the viewer window would need it's own unique name as OBS has issues refreshing to a new window with the same title (so when you select a skin and RS closes the window and opens the skin viewer)

zoggins, 2019-08-01:
My setup actually depends on them being different.  :)  It was my very first RetroSpy change.  So I will come up with a way to make us both happy

zoggins, 2019-09-16:
An option will be in place to have a static viewer window name in the next release.

closed 2019-10-03 by zoggins