retrospy/RetroSpy #86: [BUG] Settings lost if there is a bad skin

edru, 2019-08-21:
If there is a folder in the skins folder without a .xml file or with a bad value declared (ie, calling for an image that doesnt exist), you will get the warning, but upon clicking OK and RS launching, all your settings are wiped back to default.

This includes Controller Source, COM port, as well as delay (delay being the real annoying one)

edru, 2019-08-21:
Doing further testing it looks as if the com port and delay value is still intact, it just displays as null/0. It only changes value if you enter a new integer - so in this case its just the display that is bad.

zoggins, 2019-08-21:
Will look into it. Thanks

zoggins, 2019-09-15:
This will be fixed in the next release

closed 2019-10-03 by zoggins