retrospy/RetroSpy #92: Xbox 360 Startup is super picky

zoggins, 2019-09-20:


fatihG, 2022-01-15:
I just ordered a BeagleBone Black board, specifically for the 360. 

Since this was opened in 2019 and recently updated back in December, will this be 'fixed'?

The instructions say that it works '90%' of the time if you follow the startup sequence. 
But was just wondering if there was work planned to make this more seamless.

zoggins, 2022-01-15:
It is on my list to revisit soon.  Just need to get some time to dive in.

zoggins, 2023-01-30:
Actually I am hoping to release a version with improved 360 support in the next couple of days.  Its still not perfect, but its less of a pain in the butt to get going.  Currently ethernet is the only way because there is too much data for the serial communication to handle.

TheDrinkingBird, 2023-01-30:
Thank you for your informative and speedy reply! Hope things go well.

zoggins, 2023-02-01:
I am going to release the update tomorrow.  The  beaglebone will require a reimage and the new hostname will be retrospy.local.  The way to get the 360 to work is start everything up, but leave the controller unplugged.  Once the view window comes up (you are looking at the image of the controller) plug in the controller and do not touch any of the buttons or stick until the green light is solid on the controller.  If the controller does work on the Xbox, close the view window and unplug the controller.  Try again.  This can take 4 or 5 times, which sucks, but this doesn't require all the get lucky timing the old method had.\

TheDrinkingBird, 2023-02-01:
I know this has to be very hard and tedious, so thank you for all of your work!

TheDrinkingBird, 2023-02-08:
Hey Zoggins, I reimaged my sd card, reflashed the BB, followed the instructions in your last comment to the best of my understanding, yet I can't seem to get it working. 1. plug in BB USB & xbox 360 2. start up retro spy (hostname: retrospy.local) & hit go 3. wait for input view window (picture of controller appears) then plug in controller 4. don't press a button until the green light has stopped blinking. Is that correct?

zoggins, 2023-02-08:
You tried 4 or 5 times and still nothing? What controller and console version?

zoggins, 2023-02-08:
Also what the symptoms? RetroSpy shows inputs, but no controller activity on the Xbox?

TheDrinkingBird, 2023-02-09:
I've tried at least 20 times. I have a white wired xbox 360 controller for windows (I customized its appearance, but no modifications), I'll try to send a picture. It's a black xbox s. Console Serial Number: 068340703808 & Console ID: 026791782282. I will also send a picture of the xbox. My symptoms: RetroSpy shows inputs no problem, but no controller activity on xbox. Thank you for the help again! 

[image]

[image]

[image]

[image]

[image]

zoggins, 2023-02-09:
I have two of the same Xboxs, and I just got a black OEM controller.  I have been testing on a 3rd party controller.  I will see if I have the same problem with the new controller.

zoggins, 2023-02-09:
Also what version of the UI are you using and on what OS?

zoggins, 2023-02-09:
Also is the Xbox 360 modded at all? I have one modded and one not modded, just trying to simulate the best I can.

TheDrinkingBird, 2023-02-09:
My Xbox is Not modded at all. xbox Dashboard: 2.0.17559.0 & Windows 10 Home. Version: 21H2. OS Build: 19044.2486. 64-bit operating system.

[image]

zoggins, 2023-02-09:
This is a video of my process for getting working on my 360.

https://retrospycom-my.sharepoint.com/:v:/g/personal/zoggins_retro-spy_com/EUjbfO7MJPZDvFUmsbu47MUB6Dv8V2wDBHET97zE73Dknw?e=OMC5L7

TheDrinkingBird, 2023-02-10:
[https://youtu.be/1DjVZyla61w](url)

TheDrinkingBird, 2023-02-10:
https://youtu.be/1DjVZyla61w

zoggins, 2023-02-10:
Let me put together a debugging program that will let me see where it is failing.  I am hoping I can get it done tonight, but might be tomorrow morning.  Sorry for the headache.

TheDrinkingBird, 2023-02-10:
No worries! I can't wait to find out it was me doing one tiny thing wrong this whole time!

zoggins, 2023-02-10:
Inside the zip file (linked) run UsbUpdater.exe instead of RetroSpy.  Once it connects plug in the controller.  Copy the output and send it my way.  Its a little hacky, sorry.

https://retrospycom-my.sharepoint.com/:u:/g/personal/zoggins_retro-spy_com/ERRKuSjf22BMkDhZk6a4uiwBvpKYPXjvNoDL0AaAwg9SYQ?e=3lvIYY

TheDrinkingBird, 2023-02-10:
Plug in the controller to the BeagleBone? do I need to turn on the xbox? Sorry if the file explains itself, haven't downloaded yet

zoggins, 2023-02-10:
Do everything like you normally, just use usbupdat er in the zip file instead of retrospy

TheDrinkingBird, 2023-02-10:
[RetroSpy USB Updater.txt](https://github.com/retrospy/RetroSpy/files/10708637/RetroSpy.USB.Updater.txt)

zoggins, 2023-02-10:
Do you have any other USB devices plugged into the 360?

TheDrinkingBird, 2023-02-10:
No, only the beaglebone

zoggins, 2023-02-10:
The xbox is connected to the network via an ethernet cable?  Or not connected at all?  Just trying figure out why we have completely different behaviors for the same setup, lol.

TheDrinkingBird, 2023-02-10:
My Xbox is Not connected with an ethernet cable

zoggins, 2023-02-10:
Can I see the label of the back of your controller?

zoggins, 2023-02-10:
Also what is the MFG date on the back of the Xbox?

zoggins, 2023-02-10:
Re-download the zip file and run UsbUpdater with the controller already plugged in and send the output.

TheDrinkingBird, 2023-02-10:
Lmk if these pictures don't come through. I'm about to try your last message

[image]

[image]

[image]

[image]

TheDrinkingBird, 2023-02-10:
It gets stuck on (Not Responding). I re-downloaded the zip file and plugged in the controller before I ran the program 

[image]

zoggins, 2023-02-10:
hmmm, that isn't what I wanted.  let me send a new link

zoggins, 2023-02-10:
https://retrospycom-my.sharepoint.com/:u:/g/personal/zoggins_retro-spy_com/EZXi32nOCadIiUasJx4nYpQB3hjkNpwnHZfWBbv8yyhOyg?e=PLYnav

TheDrinkingBird, 2023-02-10:
I don't see USB Updater in this new Zip file, run on retrospy?

zoggins, 2023-02-10:
there isn't a usbupdater.exe?

TheDrinkingBird, 2023-02-10:
[image]

[image]

[image]

[image]

TheDrinkingBird, 2023-02-10:
WinRar gave me this message, so I used BreeZip to extract it.

zoggins, 2023-02-10:
https://retrospycom-my.sharepoint.com/:u:/g/personal/zoggins_retro-spy_com/EaGFdiizhvpMo88TJxQhwHMBkLrD707owTtPtkfZ6SeKZQ?e=5goVej

TheDrinkingBird, 2023-02-11:
I'll get back to you tomorrow morning with my results

TheDrinkingBird, 2023-02-11:
[image]

[image]

TheDrinkingBird, 2023-02-11:
It gives me this message when I try to run UsbUpdater. Downloaded .NET Desktop Runtime, still won't open

TheDrinkingBird, 2023-02-16:
Thank you for going this in-depth, really appreciate it!

TheDrinkingBird, 2023-03-07:
Hey Zoggins, any luck with the xbox 360 controller?

zoggins, 2023-03-09:
I got the exact controller of Ebay.  It just got here, so still working on it.

zoggins, 2023-03-28:
Do you have the box your controller came in?  if so what is the serial number on the bottom of the box?

zoggins, 2023-03-29:
I had a breakthrough.  I think I have it working doing some more testing.  The long story short is that I am pretty sure these controllers are not 100% legit and they some dodgy stuff around the controller authentication process, so I had to kinda patch up the authentication process.

zoggins, 2023-03-29:
I have checked in the change.  Its not in the image yet.  The Vision USB updater should pull it down though if you want it early.

TheDrinkingBird, 2023-03-30:
[image]

TheDrinkingBird, 2023-03-30:
lmk if you can't see the picture,

TheDrinkingBird, 2023-03-30:
[image]

zoggins, 2023-09-20:
This should be fixed in the new firmware

closed 2023-09-20 by zoggins