retrospy/RetroSpy #99: 8BitDo SN30 Pro USB becomes unusable over time

mn1213, 2019-10-21:
8BitDo SN30 Pro USB (model 82AA) -> Beaglebone Green -> Nintendo Switch

Inputs start out buttery smooth, but become increasingly delayed as time passes. usb-mitm's memory use also climbs as a buffer somewhere appears to get clogged. A similar *symptom* was documented at https://github.com/usb-tools/USBProxy-legacy/issues/60 in the past.

Repro is simple: Start the proxy and press a few buttons to ensure everything is working as expected. Leave it alone for several minutes and press a few more. It will take seconds to see the result.

zoggins, 2019-10-21:
> usb-tools/USBProxy-legacy#60

will take a look

zoggins, 2019-10-21:
My guess is the conversion talked about in that bug isn't right for this
controller.  I will see what it looks like when I get home.

zoggins, 2019-10-21:
You are seeing this from RetroSpy or in the terminal on the beaglebone?

zoggins, 2019-10-21:
Also have tried with any other types of controllers. or just this one?

zoggins, 2019-10-21:
I can repro on the SN30, it takes a little over 5 minutes to fall over.  It
doesn't seem to happen on a pro controller, or takes longer. Hard to tell.
Memory is leaking somewhere in all cases, but memory usage is double when
using the SN30 versus a pro controller.

mn1213, 2019-10-22:
I don't have any other supported controllers to test, nor do I have a modified Switch unfortunately. 

Although I'm real under-equipped for information gathering here, I can at least add that no amount of fiddling with the order of operations or use of USB hubs in the chain have any positive effect on the situation.

Thinking back to that issue I linked (though it's probably not what we're seeing here), how do the speed and polling rate of a real Pro Controller compare to an SN30 Pro and to the root hub?

zoggins, 2019-10-22:
I have a theory what is happening, but I need to test it.  The SN30 is
really an XInput controller that can fake being a Switch Pro controller.
When in switch mode it sends an identical descriptor as the Pro controller
including th polling interval, but I suspect even thought it is
identifying its polling interval as the same as the Pro its actually still
operating with the XInput polling rate which is much faster, like 1 instead
of 10.  I should be able to test the theory today or tomorrow.

mn1213, 2019-10-22:
That seems like a sound theory, and there are definitely firmware quirks at play here. I can't find a single program that supports DirectInput or Nintendo controllers which can properly connect to and/or read from the controller.

zoggins, 2019-10-26:
i think i have a fix.  I will try to have a test image out to tomorrow

zoggins, 2019-10-26:
Here is some test bits to try.  Sometimes you may have to try a few times to get everything to connect, but i think its pretty solid.
http://www.zoggins.net/files/9c769400-0838-44cc-92e1-fbdb5bfd4523/BBB-blank-debian-9.9-retrospy-iot-armhf-2019-10-25-4gb.img.xz
http://www.zoggins.net/files/9c769400-0838-44cc-92e1-fbdb5bfd4523/RetroSpy-release.zip

mn1213, 2019-10-28:
I just tested the new image, though my Windows machine is out at the shop for repairs, so I've not tested the client. The results were quite opposite of what was expected, with inputs taking seconds to process at the time of initial connection. I see you've turned on some extra logging, so here's what I gathered:

```
$ sudo usb-mitm  
Loading plugins from /usr/local/lib/USBProxy/
vendorId=ffffffff
productId=ffffffff
cleaning up /tmp
removing 1
Made directory /tmp/gadget-3mmef3 for gadget
Printing Config data
	Strings: 2
		DeviceProxy: DeviceProxy_LibUSB
		HostProxy: HostProxy_GadgetFS
	Vectors: 0
Pointer: 0
  Manufacturer: Nintendo Co., Ltd.
  Product:      Pro Controller
  Serial:       000000000001
	*Config(1): 09 02 29 00 01 01 00 a0 fa
		Interface(0):
			*Alt(0): 09 04 00 00 02 03 00 00 00
				HID: 09 21 11 01 00 01 22 cb 00
				EP(81): 07 05 81 03 40 00 02
				EP(02): 07 05 02 03 40 00 08
[81]: old bInterval: 02
[05]: calculated new bInterval: 81
[02]: old bInterval: 08
[07]: calculated new bInterval: 02
searching in [/tmp/gadget-3mmef3]
Starting setup writer thread (5323) for EP00.
Starting setup reader thread (5322) for EP00.
Opened EP81
Opened EP02
Starting writer thread (5330) for EP02.
Starting reader thread (5329) for EP02.
Starting writer thread (5328) for EP81.
Starting reader thread (5327) for EP81.

Warning: queue fills up! Feel free to search the bug in either the driver or usbproxy.
```

```
top - 01:38:02 up 6 min,  2 users,  load average: 1.77, 1.44, 0.76
Tasks:  91 total,   1 running,  62 sleeping,   0 stopped,   0 zombie
%Cpu(s): 41.7 us, 27.8 sy,  0.0 ni, 29.0 id,  0.1 wa,  0.0 hi,  1.4 si,  0.0 st
KiB Mem :   494888 total,   337956 free,    70556 used,    86376 buff/cache
KiB Swap:        0 total,        0 free,        0 used.   407508 avail Mem 

  PID USER      PR  NI    VIRT    RES    SHR S %CPU %MEM     TIME+ COMMAND     
 5318 root      20   0  100532   7692   2724 S 72.7  1.6   2:23.38 usb-mitm    
25308 root      20   0    7188   2852   2328 R 13.6  0.6   0:00.08 top         
    8 root      20   0       0      0      0 I  4.5  0.0   0:04.41 rcu_preempt 
    1 root      20   0   25524   4984   3872 S  0.0  1.0   0:03.10 systemd  
```

Is there perhaps some magic in the client that I'm missing out on?

zoggins, 2019-10-28:
try "sudo usb-mitm -z"  I had to put some special handling in specifically for switch.

mn1213, 2019-10-28:
The result with `sudo usb-mitm -z` is unfortunately the same. I'll see if I can roll back the controller's firmware later today for a little extra testing just in case.

```
$ sudo usb-mitm -z
Loading plugins from /usr/local/lib/USBProxy/
vendorId=ffffffff
productId=ffffffff
cleaning up /tmp
removing 1
Made directory /tmp/gadget-ayyLdk for gadget
Printing Config data
	Strings: 2
		DeviceProxy: DeviceProxy_LibUSB
		HostProxy: HostProxy_Switch
	Vectors: 1
		Plugins:
			PacketFilter_Switch
Pointer: 1
		PacketFilter_Switch::file: 0xb6d73d60
  Manufacturer: Nintendo Co., Ltd.
  Product:      Pro Controller
  Serial:       000000000001
	*Config(1): 09 02 29 00 01 01 00 a0 fa
		Interface(0):
			*Alt(0): 09 04 00 00 02 03 00 00 00
				HID: 09 21 11 01 00 01 22 cb 00
				EP(81): 07 05 81 03 40 00 02
				EP(02): 07 05 02 03 40 00 08
[81]: old bInterval: 02
[05]: calculated new bInterval: 81
[02]: old bInterval: 08
[07]: calculated new bInterval: 02
searching in [/tmp/gadget-ayyLdk]
Starting setup writer thread (1447) for EP00.
Starting setup reader thread (1446) for EP00.
Opened EP81
Opened EP02
Starting writer thread (1454) for EP02.
Starting reader thread (1453) for EP02.
Starting writer thread (1452) for EP81.
Starting reader thread (1451) for EP81.
00000000000000000000000001111111111111110111111111111111
11000000100111001110000011110000101011000001000111111110
11000000100111001110000011110000101011000001000111111110
00000000000000000000000001111111111111110111111111111111
00000000000000000000000001111111111111110111111111111111
00000000000000000000000001111111111111110111111111111111
00000000000000000000000011111110111111101111111011111110
00000000000000000000000001111111111111110111111111111111
00000000000000000000000000111111000111111101111100101111
00000000000000000000000001111111111111110111111111111111
00000000000000000000000011111110111111101111111011111110
00000000000000000000000010111111111011111001111110101111
00000000000000000000000001111111111111110111111111111111
00000000000000000000000000111111000111111101111100101111
00000000000000000000000000111111000111111101111100101111
00000000000000000000000001111111111111110111111111111111
00000000000000000000000001111111101011111001111100101111
00000000000000000000000001111111111111110111111111111111
00000000000000000000000001111111101011111001111100101111
00000000000000000000000001111111111111110111111111111111
00000000000000010000000000111111111101111000000001111111
00000000000000000000000001111111111111110111111111111111
00000000000000010000000000111111111101111000000001111111
00000000000000000000000001111111111111110111111111111111
00000000000000010000000000111111111101111000000001111111
00000000000000000000000001111111111111110111111111111111
00000000000000000000000000111111000111111101111100101111
00000000000000000000000001111111111111110111111111111111
00000000000000000000000010111111101011111001111111001111
00000000000000000000000001111111111111110111111111111111
00000000000000000000000001111111101011111001111100101111
00000000000000000000000001111111111111110111111111111111
00000000000000000000000001111111101011111001111100101111
00000000000000000000000001111111111111110111111111111111
00000000000000000000000001111111101011111001111100101111
00000000000000000000000001111111111111110111111111111111
00000000000000000000000001111111101011111001111100101111
00000000000000000000000001111111111111110111111111111111
00000000000000000000000001111111101011111001111100101111
00000000000000000000000001111111111111110111111111111111
00000000000000000000000001111111111111110111111111111111
00000000000000000000000000111111111011111001111100101111
00000000000000000000000001111111111111110111111111111111
...
00000000000000000000000001111111111111110111111111111111
00000000000000000000000001111111111111110111111111111111
```

zoggins, 2019-10-28:
I will double check on my end, but I had this working with usb-mitm -z.

zoggins, 2019-10-28:
I am assuming you let it run for a little bit?  The slowdown when it first
starts is normal, but then it should catch up.  But it should catch up
pretty fast.

zoggins, 2019-10-28:
Just for clarity the immediate slowdown is a side effect of fixing the memory leak.  So that part is expected, but its supposed to heal itself after a handful of seconds which is what i will look at when I get home from work.

mn1213, 2019-10-28:
That's right, the situation continues to worsen the longer the proxy is running.

zoggins, 2019-10-28:
And you are doing this all over SSH?

zoggins, 2019-10-28:
Is this a Green wired or wireless?  We now have a mystery.  I can't repro with the latest image on a Green Wired.  So we have to figure out where the difference between my setup and your setup is.

mn1213, 2019-10-29:
Yup, it's a Green Wired that I'm connecting to over SSH. Thinking back to my other issue, we know the label on my controller and instruction manual are different from some others - failing to mention that holding Y boots the controller in DirectInput mode - so my best guess is that it's either a hardware or firmware difference.

The model number on my sticker is 82AA, my manual looks like https://download.8bitdo.com/Manual/Controller/SN30pro-USB/SN30-Pro-USB.pdf, and I'm on the 1.03 firmware found at https://download.8bitdo.com/Firmware/Controller/SN30pro-USB/ (which can be installed by connecting the controller to your PC while holding D-pad Up and extracting the ZIP to the mass storage device that appears).

zoggins, 2019-10-29:
I will verify my info when I get home

zoggins, 2019-10-29:
> That is definitely the same manual I have.  I have no sticker with the
> model on my controller and I will check the firmware when I get home.
>
> On Tue, Oct 29, 2019 at 8:11 AM Christopher Mallery <

Could you give me the startup output of sudo usb-mitm -ldddd?

zoggins, 2019-10-30:
So controller seems to be a newer version that the firmware cannot be updated on and its currently at 1.02.  Could be the firmware difference or the difference in version of the controller. Hmmm....what next...

zoggins, 2019-10-30:
I am winging it on this one, but lets try this.  Login as debian and then:

cd USBProxy/src/build
git pull
git checkout remove_duplicate_code
chmod 755 ../tools/git-version-gen.sh
sudo make install

then give "sudo usb-mitm -z" a whirl.  I am not convinced this will help, but its a test.

mn1213, 2019-11-01:
Not terribly important, but it seems I remembered the firmware update procedure all wrong.
1. Extract BIN from firmware ZIP
2. Hold L1+R1 and connect controller (for mass storage mode)
3. Copy BIN to controller's storage
4. Press D-pad Up to commit the firmware and reboot

---

Back on toppic, I dare say you nailed it with the latest commit to the remove_duplicate_code branch. CPU utilization is way down, memory allocation is stable, the controller feels nice and responsive, and it wraps quite nicely inside websocketd for browser-based input logging.

Once again you've been awesome, thanks so much for the help!

closed 2019-11-01 by mn1213

zoggins, 2019-11-01:
Always glad to help!