retrospy/RetroSpy #103: Intermittent Crash/Hang on Display Start

FerretBomb, 2019-11-25:
Intermittently RetroSpy will crash after selecting port/controller/skin and hitting "Go!". The display window is never shown, but the config window vanishes.
While the process shows in Task Manager, it cannot be ended and gives an 'Access Denied' error, even when running with Administrator escalation.
Likewise, tasklist shows the RetroSpy.x64.exe process, but when taskkill /F /PID <pid> is used, it returns 'no running instance of the task'.
Also-likewise, wmic terminate does not end the process.
Attempting to restart RetroSpy on the correct COM port errors out that access is denied (but can be started on a different COM port without issue, and without connecting to the inline).

SOMETIMES, unplugging the inline and re-plugging it will allow a new run of RetroSpy to connect to the correct COM port again. The old task still shows in the task-list however, and still is unkillable.
Most of the time, a reboot is required to free up the COM port and re-establish connection. Always requires a reboot to get rid of the old 'dead' retrospy instance from the task list.

Unfortunately I am unable to reliably reproduce the issue so cannot test possible fixes.
Random guess, could OBS's window capture be jumping in at just the wrong moment when the new display window is being constructed? I believe that each of the 3-4 times I have had this happen, OBS was already open at the time. I have not yet had it occur when OBS was not running, but that may be just down to luck, that generally when I'm using RS it's for a stream, and I tend to open OBS first in my setup checklist.

zoggins, 2021-12-02:
Is this still a problem?

FerretBomb, 2021-12-03:
Cannot confirm or dismiss as fixed as I have not used RetroSpy in the last year, and the issue was intermittent when it occurred. Can assume-fixed and if it recurs I will reach out again with as much information as I am able. Thank you!

closed 2021-12-03 by FerretBomb