retrospy/RetroSpy #104: Retrobit 2.4Ghz Saturn controller not registering properly in Retrospy

Shentokk, 2019-12-03:
I am unable to have the Retrobit Saturn wireless controller display properly in Retrospy. It works fine in giving commands to the Saturn, but the Retrospy application shows it pressing random buttons and constantly pressing Up/Down simultaneously. I updated to the latest release as of this post (v3.5) to see if there was a fix, but the issue still occurs. Please let me know if there is any other information I can provide.
[image]

zoggins, 2019-12-03:
This one?
https://www.amazon.com/Retro-Bit-Official-Wireless-Controller-8-Button/dp/B07XVHSNCS/ref=pd_sbs_147_t_0/142-7910219-6207636?_encoding=UTF8&pd_rd_i=B07XVHSNCS&pd_rd_r=240dcba3-5980-4078-b412-dd3d25a15a34&pd_rd_w=pN2SA&pd_rd_wg=pS2j9&pf_rd_p=5cfcfe89-300f-47d2-b1ad-a4e27203a02a&pf_rd_r=809JS3PGF89DEF863ZJB&psc=1&refRID=809JS3PGF89DEF863ZJB

I will get one and see, its not shocking that it doesn't work correctly.
3rd party controllers almost always have idiosyncrasies.

zoggins, 2019-12-03:
Which firmware have you tried?  There are 2 saturn firmwares.  The one for
digital controllers and the one for the 3D controller.

Shentokk, 2019-12-03:
Yeah, that controller. I'm using the digital firmware.

zoggins, 2019-12-04:
I have a fix, it does require a firmware update though.  The timings are different for the controller than a standard one, so I had to compensate for that in the firmware.

closed 2020-05-16 by zoggins