retrospy/RetroSpy #125: Signal noise on SNES when CPU is working

FineWolf, 2020-05-05:
**Board:** Arduino Uno
**Firmware Version:** v3.6 in MODE_SNES

I've been having an issue where, during load transitions in game, random shadow inputs are being reported by RetroSpy. During normal gameplay however, this issue isn't present at all.

When I was looking to diagnose the issue, I of course defined `DEBUG` in order to get a clean output. Turns out, that solved the issue (while increasing latency). Turns out the modulo operation added enough latency between reads to solve the issue. 

I've modified the code in order to add the additional modulo operation in the regular output, but it does add some latency between input press and detection.

Here's a video of the issue: https://youtu.be/Pw7qoQNSgrU

Is there a time gate that needs to be added in order to prevent false LATCH and CLOCK signals to be interpreted as actual signals or is there an issue with my wiring (which would be weird considering it only happens during CPU/PPU intensive workloads or level transitions)? Any thoughts?

zoggins, 2020-05-05:
Do you have a the controller ground connected to Arduino ground?

FineWolf, 2020-05-05:
Yes, the ground is connected (and verified using a continuity tester) to the ground line of the SNES controller and the GND pin of the Arduino Uno.

zoggins, 2020-05-05:
Hmmm, let me think about it.  I will hook the logic analyzer up the SNES
and see if I can see any noise that your Arduino might be picking up.  I
have never actually analyzed an SNES cable.

FineWolf, 2020-05-05:
What I don't understand however is why the added modulo is affecting the output. I should still theoretically pick up noise if it was a noise issue.

FineWolf, 2020-05-06:
@zoggins After hitting my head against the wall multiple times and hooking up the leads to a logic analyzer, my original guess was right. This wasn't at all an electrical problem.

`sendRawData` is being improperly optimized by gcc causing weird behavior. Adding the branch just prevented `gcc` from doing the erroneous optimization, which is why the behavior went away when `DEBUG` was active.

The solution is to wrap the method in pragma statements as such:

```
#pragma GCC optimize ("-O1")
#pragma GCC push_options

/////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////
// Sends a packet of controller data over the Arduino serial interface.
inline void sendRawData( unsigned char first, unsigned char count )
{
 // ...
}
#pragma GCC pop_options
```

This completely removed the issue.

zoggins, 2020-05-06:
Cool.  I will add it to master and it will go out in the next release.
Thanks for the heads up!

closed 2020-05-23 by zoggins