retrospy/RetroSpy #134: Updating MODE_DETECT Tutorial

Shentokk, 2020-06-02:
Will there be plans to update the MODE_DETECT tutorial to work with the new codebase? I tried doing what I did before and just got an error with compiling. This is with the 3.7 release. Maybe there's just something I'm not understanding with the code change.

```Arduino: 1.8.12 (Windows Store 1.8.33.0) (Windows 10), Board: "Arduino Uno"

C:\RetroSpy-release\firmware\firmware.ino: In function 'void setup()':

firmware:188:18: error: expected unqualified-id before '.' token

         NeoGeoSpy.setup();

                  ^

firmware:190:19: error: expected unqualified-id before '.' token

         GenesisSpy.setup();

                   ^

firmware:192:18: error: expected unqualified-id before '.' token

         SaturnSpy.setup();

                  ^

firmware:194:18: error: expected unqualified-id before '.' token

       Saturn3DSpy.setup();

                  ^

firmware:203:5: error: 'else' without a previous 'if'

     else {

     ^~~~

firmware:204:20: error: expected unqualified-id before '.' token

         Saturn3DSpy.setup();

                    ^

C:\RetroSpy-release\firmware\firmware.ino: In function 'void loop()':

firmware:265:18: error: expected unqualified-id before '.' token

         NeoGeoSpy.loop();

                  ^

firmware:267:19: error: expected unqualified-id before '.' token

         GenesisSpy.loop();

                   ^

firmware:269:18: error: expected unqualified-id before '.' token

         SaturnSpy.loop();

                  ^

firmware:271:20: error: expected unqualified-id before '.' token

         Saturn3DSpy.loop();

                    ^

firmware:280:2: error: 'else' without a previous 'if'

  else {

  ^~~~

firmware:281:20: error: expected unqualified-id before '.' token

         Saturn3DSpy.loop();

                    ^

exit status 1
expected unqualified-id before '.' token

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
```
This is what I have written in those sections:

```#elif defined(MODE_DETECT)
    if ( !PINC_READ(MODEPIN_SNES)) {
        NeoGeoSpy.setup();
    } else if ( !PINC_READ(MODEPIN_N64))  {
        GenesisSpy.setup();
    } else if ( !PINC_READ(MODEPIN_GC)) {
        SaturnSpy.setup();
    } else {
        Saturn3DSpy.setup();
    }`

`#elif defined(MODE_DETECT)
    if( !PINC_READ( MODEPIN_SNES ) ) {
        NeoGeoSpy.loop();
    } else if( !PINC_READ( MODEPIN_N64 ) ) {
        GenesisSpy.loop();
    } else if( !PINC_READ( MODEPIN_GC ) ) {
        SaturnSpy.loop();
    } else {
        Saturn3DSpy.loop();
    }```

zoggins, 2020-06-02:
You have to modify this section of code:
#elif defined(MODE_DETECT)
NESSpy NESSpy;
SNESSpy SNESSpy;
N64Spy N64Spy;
GCSpy GCSpy;
DreamcastSpy DCSpy;
WiiSpy WiiSpy;
#endif

  to contain the spys you need for your setup, so in your case

#elif defined(MODE_DETECT)
NeoGeoSpy NeoGeoSpy;
GenesisSpy GenesisSpy;
SaturnSpy SaturnSpy;
Saturn3DSpy Saturn3DSpy;
#endif

I will think of a better solution.

Shentokk, 2020-06-02:
Thanks. I really appreciate the quick response! I'm just quite rusty with programming so a lot of this new stuff is a lot to take in.

zoggins, 2020-06-04:
With the next release instead of just uncommented MODE_DETECT in order to use MODE_DETECT you will need uncomment MODE_DETECT and any modes you are using.  So in your case you would uncomment MODE_DETECT, MODE_NEOGEO, MODE_GENESIS, MODE_SATURN and MODE_SATURN3D.  I will get the tutorial updated.

closed 2020-07-02 by zoggins