retrospy/RetroSpy #138: Keybinding issue on latest version

Skolig, 2020-06-12:
Tried a variety of key/button combinations but without exception when I trigger those keybinds I get the error 

`Could not load type 'InputUnion' from assembly 'Retrospy, Version=3.7.0.0, Culture=neutral, PublicKeyToken=null' because it contains an object field at offset 0 that is incorrectly aligned or overlapped by a non-object field.`

For what it's worth, my `keybindings.xml` file currently looks like

```xml
<?xml version="1.0" encoding="UTF-8"?>
<keybindings>
    <binding output-key="home">
        <input button="z" />
    </binding>
</keybindings>
```

This is for a GameCube controller. The input display is otherwise fine.

zoggins, 2020-06-12:
That is an interesting one, I will take a look.  Thanks!

zoggins, 2020-06-12:
Good news! I can reproduce the problem.  Bad news!  I have no idea what is
wrong.  I will sort it out.

zoggins, 2020-06-12:
I think I found the problem.  There is a bigger issue I have to solve, but this stop gap release should be working.

https://retro-spy.com/download/RetroSpy-release.zip

Let me know if it works or not. Thanks!

Skolig, 2020-06-12:
It works, the only issue I have now is that the delay variable also affects keybinds, ideally I'd use the keybinds to control my timer but I use the delay feature within RetroSpy to account for the input introduced by my video capture.

zoggins, 2020-06-12:
Interesting.  I am trying to think through what that would look like.  Let
me put some thought into it.

zoggins, 2020-06-12:
Its a long shot, but I just uploaded a new version that might help your problem.  Not 100% sure, but let me know.

Skolig, 2020-06-13:
Still behaving the same way; having the delay set to 1500ms causes keybindings to occur with the same delay instead of instantly.

zoggins, 2020-06-13:
Ok, I will figure something out.

zoggins, 2020-06-13:
It seems to be working for me. Maybe my upload was bad.  Let me upload it to github

zoggins, 2020-06-13:
Yes it appears the old zip file is cached.  I will get it uncached after dinner, but on my end it appears to be working.

zoggins, 2020-06-13:
https://github.com/retrospy/RetroSpy/releases/tag/v3.7.1

Try that.

Skolig, 2020-06-13:
Yep, everything seems to be working perfectly now. Thanks!

zoggins, 2020-06-13:
Awesome.  Glad I could help.  Let me know if you need anything else. Thanks!

closed 2020-07-02 by zoggins