retrospy/RetroSpy #147: 8BitDo Wireless USB Adapter Does Not Work

ToransuShoujo, 2020-07-01:
Hello,

I purchased the 8BitDo Wireless USB Adapter because it has support for HD rumble and no input lag on Pro Controllers (as opposed to USB connection, which does introduce input lag). However, the adapter does not work with RetroSpy. I find this strange as using the Mayflash Magic-NS does actually work.

I contacted 8BitDo to get a firmware to force the adapter into Switch mode, as the device first connects to the BeagleBone in X-Input mode. Even with this, however, RetroSpy sees no inputs and the controller does not work. This device does work when connected straight to the Switch dock.

I'm not sure what to do from this point. Any help would be appreciated. Thank you so much.

zoggins, 2020-07-01:
Is it this adapter?
https://www.amazon.com/Wireless-Bluetooth-Adapter-Nintendo-Switch-Raspberry/dp/B0786JC6VW/ref=sr_1_2?m=A2UWVAXOHJ0A2X&marketplaceID=ATVPDKIKX0DER

I will take a look at the problem.  Thanks!

zoggins, 2020-07-01:
Also do you have a link or could you attach the firmware to force it into switch mode?

ToransuShoujo, 2020-07-01:
It is that adapter, yes. I personally use the gray one (the one for PlayStation Classic), but they are essentially the same. 

I've attached the file, which contains the firmware itself and a readme for instructions. Thanks again for your help.
[USBRR131NS.zip](https://github.com/retrospy/RetroSpy/files/4859506/USBRR131NS.zip)

zoggins, 2020-07-01:
Glad to help.  Adapter is on its way and will look into it as soon as I get it!

zoggins, 2020-07-02:
I forgot to ask.  What controller are you pairing with the adapter?

ToransuShoujo, 2020-07-02:
I'm using the Nintendo Switch Pro Controller with the adapter.

Only reason is because connecting the Pro Controller via USB directly introduces a bit of input lag.

zoggins, 2020-07-02:
I can't get my Switch Pro controller/Switch to work with the grey adapter.  It syncs with the controller, but nothing happens on the Switch.

zoggins, 2020-07-02:
This is without RetroSpy in the middle, I must be doing something wrong.

ToransuShoujo, 2020-07-02:
I think if you're using the firmware I gave you, you need to manually switch the mode on the adapter as indicated in the README. If not, perhaps updating the firmware will work?

zoggins, 2020-07-02:
Yup figured it out, had to figure out - was select :)

zoggins, 2020-07-02:
Definitely a problem with the firmware not playing nicely with the adapter.  I am working on it.

ToransuShoujo, 2020-07-09:
Hello there! I hope this message doesn't bug you, but I wanted to ask if there was any news regarding this issue.

Thank you again for your hard work!

zoggins, 2020-07-09:
I am working on it.  Basically the adapter is restarting during initialization and I can't figure out why.

Sent from Nine<http://www.9folders.com/>

ToransuShoujo, 2020-07-09:
Ah, interesting. Thank you again.

zoggins, 2020-07-09:
I am absolutely working on the issue, but it’s a bit of a mystery at the moment.  Its weird  that then SN30 controller, which appears to have the same USB communication with the Switch, works and the adapter doesn’t.

zoggins, 2020-07-10:
Current update.  The problem is that the Switch doesn't think anything is connected to it, even though it seems to be talking to the controller during all the handshaking.  So the Switch sends a reset to the adapter looking for something alive.  I am still trying to find a solution.

ToransuShoujo, 2020-07-10:
Thank you again for looking into this, I appreciate it so much.

ToransuShoujo, 2020-07-15:
I was thinking about this and I had a couple of ideas, although I don't know if they are of any use.

I know the Magic-NS from Mayflash does work, but I'm not sure what is different about it. Maybe it is possible to compare them?

I also know that there was something along the lines of fixing the restart issue in the 8BitDo changelogs, but it didn't have support for swapping to Switch mode. Maybe it is possible to trick it into Switch mode somehow, instead of X-input?

Sorry if these are unhelpful.

zoggins, 2020-07-16:
Currently I am comparing it to the SN30 pro from 8bitdo that works fine.  Not sure what the difference is though.