retrospy/RetroSpy #149: Nintendo SNES

810munoz, 2020-07-10:
Clean install of RetroSpy. Doesn't detect L, R, X, or A on Nintendo SNES input display.

((B, Y, Start, Select, and D-Pad all detect normally))

zoggins, 2020-07-10:
Using 3.8 firmware and exe?  What skin are you using?  What you are describing is if there is NES firmware on your Arduino and you are trying to use it on an SNES.  I will double check, but I just test SNES myself and it was working fine.

810munoz, 2020-07-10:
Using default Nintendo SNES skin with ihaveonebigeye's input display set to SNES.

810munoz, 2020-07-10:
And you're right, it behaves the same way if I have it on SNES or NES setting. hmm...

zoggins, 2020-07-10:
Nothing is jumping out, but I am looking into this.

810munoz, 2020-07-10:
I guess it could just be faulty hardware. I talked to ihaveonebigeye and he says he'll get started on a replacement. It's pretty weird since he tested it before shipping, but we'll see. Thanks for the super fast replies!

zoggins, 2020-07-10:
Closing for now. Let me know if you have any other problems.

closed 2020-07-10 by zoggins