retrospy/RetroSpy #158: RetroSpy for Switch not reading inputs

PrinceKomali, 2020-08-14:
Hi, me again. 
So i'm helping someone set up RetroSpy using the beaglebone green wireless method, and so far it hasn't successfully worked yet (we have tried literally everything). All its doing is having random buttons held down (see the attached 
[image]). Any ideas? I've tried all the steps with my beaglebone/RetroSpy and it works fine

zoggins, 2020-08-14:
First, what controller are you trying to use?

[file]<https://user-images.githubusercontent.com/37776664/90285441-89863e80-de39-11ea-96ef-fede6eabb3d9.png>). Any ideas? I've tried all the steps with my beaglebone/RetroSpy and it works fine

PrinceKomali, 2020-08-14:
Official switch pro iirc

zoggins, 2020-08-14:
Does the control function on the console?  Or is there no inputs to the console?

PrinceKomali, 2020-08-14:
I'm not sure what you mean by console

zoggins, 2020-08-14:
Can you navigate the Switch’s menu with RetroSpy running?  Even with the display messed up, is the controller able to communicate to the Switch.

So plug the controller into the Beaglebone, plug the Beaglebone into the Switch Dock, Fire up RetroSpy.  As you have shown the display will be messed up, but does the controller successfully navigate the Switch menus?

I am just trying to narrow down where the problem is. As it could be a problem with communicating with the beaglebone communicating with the switch or a problem with the beaglebone communicating with the RetroSpy UI.

zoggins, 2020-08-14:
Also is this the new RetroSpy firmware and UI?

PrinceKomali, 2020-08-14:
Sorry I edited my comment, and I guess the email didn't update!
Yes he can navigate the menu and it is the latest RetroSpy version and beaglebone image
when I first saw "console" i thought you meant command prompt :/

zoggins, 2020-08-14:
I will reflash my Green Wireless to see if I can repro. What you describe would happen if the controller was something with a different data stream and I was incorrectly reading it, but if it’s a switch pro controller that shouldn’t be the case.   The setting for “Use Pro Controller over Wired” (or whatever it is) is on on the switch?

PrinceKomali, 2020-08-14:
uhhhh ok so we forgot to enable wired connections

really sorry for making you go through all this trouble for something so dumb

zoggins, 2020-08-14:
No worries, glad I could help.  Never worry about reaching out to me, even if it ends up being something simple.

closed 2020-08-14 by PrinceKomali