retrospy/RetroSpy #160: Settings are lost on upgrade

edru, 2020-08-18:
During the upgrade to a new version it asks to uninstall the old version. When doing so it does leave all your custom skins in place, but it removes any settings you had so that when you re-launch after the upgrade you have to reconfigure it all again. 

It forgets:
what device you had selected
the port you selected, the skin you selected
which systems were set to be excluded from the view.
Possibly the delay time
Possibly hostname for devices that require IP

edru, 2020-08-18:
It seems that the uninstall during upgrade deleted the skin.ini file for my active skin too.

zoggins, 2020-08-18:
Yeah,  that is a problem. ☹. Will take a look.

zoggins, 2021-12-01:
This has been fixed.

closed 2021-12-01 by zoggins