retrospy/RetroSpy #161: Is beaglebone a must for usb controllers?

KuribohG, 2020-08-18:
I have a raspberry Pi 3 which has enough ports for hooking up. Can we use another single board computer (like raspberry pi) instead of beaglebone?

zoggins, 2020-08-18:
The trick is it has to have both a USB client and host ports, which most Raspberry PIs don't have.  Typically they have only host ports where the Pi is acting as the USB host for things like keyboards, mice and controllers.  The client port allows the single board computer to act as a device to another USB host.  There are Pi boards out there with client ports, but I have not tested with any of them.

KuribohG, 2020-09-02:
Thanks for your reply!
If I have another type of single board computer, which has both USB client and host ports, how can I build a system image for that board? Currently we only have the image for BBB or BBG, and I haven't seen any documentation on this.

zoggins, 2020-09-02:
What computer board?  I am willing to help get the software working on another board, but I honestly have no clue if it work or how much work it would be.  The  underlying software that runs on the beaglebone is http://GitHub.com/retrospy/usbproxy the first step would be getting that to compile.

KuribohG, 2020-09-02:
Very useful information!
I am using a Zynq-7000 board called Alinx AX7010, there may be few English resource for this board.
It is an ARM + FPGA board and I know it is strange to get retrospy work on it, but this is the only board being available currently. I will take a try.

KuribohG, 2020-09-03:
I successfully built USBProxy on my board. However, when I start usb-mitm, it shows this:
```
Loading plugins from /usr/local/lib/USBProxy/
error opening library /usr/local/lib/USBProxy/DeviceProxy_LibUSB.so: cannot open shared object file: No such file or directory
Printing Config data
        Strings: 2
                DeviceProxy: DeviceProxy_LibUSB
                HostProxy: HostProxy_GadgetFS
        Vectors: 1
                Plugins:
                        PacketFilter_StreamLog
Pointer: 1
                PacketFilter_StreamLog::file: 0xb6de69f0
Segmentation fault
```
Are there any solution for this?

PS: I found for file GadgetFS_static.cpp, the `-std=gnu++11` flag is missing. I manually add this to CMake generated file. So this may be a bug in CMake configuration file.

KuribohG, 2020-09-03:
Okay, I found we must have libusb-1.0 and lorcon to build.

Are there any easy way to install lorcon on debian? Apt-get seems cannot install lorcon directly.

After I installed libusb-1.0 (but WITHOUT lorcon, I didn't find a way to install it and simply ignored this), the information is:
```
Loading plugins from /usr/local/lib/USBProxy/
vendorId=ffffffff
productId=ffffffff
cleaning up /tmp
removing 0
Made directory /tmp/gadget-DinobR for gadget
Error mounting gadgetfs from [/tmp/gadget-DinobR].
Error code from mount is: [No such device]
Printing Config data
        Strings: 2
                DeviceProxy: DeviceProxy_LibUSB
                HostProxy: HostProxy_GadgetFS
        Vectors: 1
                Plugins:
                        PacketFilter_StreamLog
Pointer: 1
                PacketFilter_StreamLog::file: 0xb6dac9f0
  Manufacturer: Logitech
  Product:      G102 Prodigy Gaming Mouse
  Serial:       147C39573635
        *Config(1): 09 02 3b 00 02 01 04 a0 96
           Name: U107.03_B0015
                Interface(0):
                        *Alt(0): 09 04 00 00 01 03 01 02 00
                                HID: 09 21 11 01 00 01 22 43 00
                                EP(81): 07 05 81 03 08 00 01
                Interface(1):
                        *Alt(0): 09 04 01 00 01 03 00 00 00
                                HID: 09 21 11 01 00 01 22 97 00
                                EP(82): 07 05 82 03 14 00 01
[81]: old bInterval: 01
[81]: calculated new bInterval: 04
[82]: oinput: Logitech G102 Prodigy Gaming Mouse as /devices/soc0/amba/e0002000.usb/ci_hdrc.0/usb1/1-1/1-1:1.0/0003:046D:C084.0003/input/input2
ld bInterval: 01
[82]: calculated new bInterval: 04
searching hid-generic 0003:046D:C084.0003: input: USB HID v1.11 Mouse [Logitech G102 Prodigy Gaming Mouse] on usb-ci_hdrc.0-1/input0
in [/tmp/gadget-DinobR]
/tmp/gadget-DinobR device file not found.
Error, unable to find gadget file.
Fail on open 11 Resource temporarily unavailable
input: Logitech G102 Prodigy Gaming Mouse as /devices/soc0/amba/e0002000.usb/ci_hdrc.0/usb1/1-1/1-1:1.1/0003:046D:C084.0004/input/input3
hid-generic 0003:046D:C084.0004: input: USB HID v1.11 Keyboard [Logitech G102 Prodigy Gaming Mouse] on usb-ci_hdrc.0-1/input1
GadgetFS not connected.
done
```

Is this correct?

zoggins, 2020-09-03:
That looks correct, the gadgetfs error

Made directory /tmp/gadget-DinobR for gadget

Error mounting gadgetfs from [/tmp/gadget-DinobR].

Error code from mount is: [No such device]

and

/tmp/gadget-DinobR device file not found.

Error, unable to find gadget file.

Fail on open 11 Resource temporarily unavailable

Is likely due to libcomposite kernel module being loaded, which can be checked by running lsmod.

On the beaglebone you shut this off by opening opt/scripts/boot/am335x_evm.sh

And comment out:

echo "${log} modprobe libcomposite"
modprobe libcomposite || true
if [ -d /sys/module/libcomposite ] ; then
        run_libcomposite
else
        if [ -f /sbin/depmod ] ; then
                /sbin/depmod -a
        fi
        echo "${log} ERROR: [libcomposite didn't load]"
fi

zoggins, 2020-09-03:
I should also add that gadgetfs need to be compiled into the kernel, it is by default on the beaglebone, not sure about your device.

zoggins, 2021-12-01:
Feel free to reopen if you are still working on this

closed 2021-12-01 by zoggins