retrospy/RetroSpy #164: RetroSpy has stopped working

codemann8, 2020-09-05:
It's been awhile since I've updated RetroSpy, but the newest version doesn't load at all. It says RetroSpy has stopped working and closes immediately.

zoggins, 2020-09-05:
Sorry to hear that.  Newest version?  And it immediately stops?  Did you download the installer or the .zip file?  Do you have .net framework 4.8 installed? Which windows version?  Sorry for all the questions, just want to narrow down the problem.

codemann8, 2020-09-07:
My apologies for the late reply. I found out it is crashing due to skins I had in there from previous versions. I'm not sure why it behaves differently with these but archiving them into another folder (until I find out the problem) fixes it.

zoggins, 2020-09-07:
If you send me the offending skins, i can debug whats going on.  if that would help.

zoggins, 2021-12-01:
Is this still an issue?

zoggins, 2021-12-09:
closing from inactivity.  Feel free to repopen if necessary.

closed 2021-12-09 by zoggins