retrospy/RetroSpy #165: Analog Stick and digipad are swapped with a generic pc controller

lordmaggoton, 2020-09-13:
i got a generic pc controller with a digipad. When i use the generic pc skin (the xbox controller) the inputs from the digipad are shown as inputs of the left analog stick. Can i fix this by changing the code in the xml file?

zoggins, 2020-09-15:
Sorry just seeing this.  I am not entirely sure the specific controller, but its highly likely you can change the skin.xml file for the PC controller skin.

wertercatt, 2021-07-09:
> i got a generic pc controller with a digipad. When i use the generic pc skin (the xbox controller) the inputs from the digipad are shown as inputs of the left analog stick. Can i fix this by changing the code in the xml file?

A lot of generic PC controllers send their d-pads as 'analog' input. No idea why.

40wattrange, 2021-11-17:
This can be resolved by adjusting the skin file. This is not a bug of the application.

zoggins, 2021-12-01:
If this is a still a problem what controller is it?

zoggins, 2021-12-09:
closing from inactivity.  Feel free to repopen if necessary.

closed 2021-12-09 by zoggins