retrospy/RetroSpy #166: PC Generic Gamepad needs a button 16

Watherum, 2020-10-12:
I was building a generic skin for a 3rd party xbox one style controller and I was unable to map the xbox button itself. 
This is because it is button 16. The generic skin helper also only maps out until button 15 causing the button to not be displayed. 

I did try just adding the line anyways, `<button name="b16" image="xbox_button.png" x="216" y="136" />` however that did not change anything.

Support for additional buttons, like 16 or even 16+ would be a great addition to map out controllers completely

zoggins, 2020-10-12:
I will take a look

zoggins, 2020-10-30:
The PC pad reader actually queries the controller driver for the number of buttons it has.

            for (int i = 0; i < _joystick.Capabilities.ButtonCount; ++i)
            {
                outState.SetButton("b" + i.ToString(CultureInfo.CurrentCulture), state.Buttons[i]);
            }

My Xbox One controller only reports 0-15, not 0-16.  My guess is the XInput driver doesn't pick up that button.

Watherum, 2020-10-31:
Well what's weird about this, is that if I use a different program like
Open Joystick display it picks up the button as button 16. And it agrees
with retrospy on the number of every other button too.
Like A being button 0 for example.
Perhaps there are some buttons that don't appear in this method of grabbing
the buttons?

zoggins, 2020-10-31:
Yeah, something definitely seems odd.  I will keep investigating.

zoggins, 2021-12-01:
If this is still an issue what controller was causing this?

zoggins, 2021-12-09:
closing from inactivity.  Feel free to repopen if necessary.

closed 2021-12-09 by zoggins