retrospy/RetroSpy #170: Sync issues with the Sega Saturn?

Ch0wW, 2020-12-15:
I seem to have some issues with creating a RetroSpy for the Sega Saturn. I followed [the tutorial on the website](https://retro-spy.com/wiki/saturn-on-arduino-getting-started/), made sure all the pins are connected to the correct Digital Pins reported on the website, uncommented the MODE_SATURN define and uploaded it on my Arduino Nano. 

When I press a button, nothing seems to happen on the program, although all inputs are properly sent to the console. I tried with 2 controller and my Arcade stick, and same error.

I've tried to use older versions of RetroSpy aswell as its firmware and when using it, random inputs occur on the program (up+left+down + A + R for instance). I don't really know if there's an error on my end, an error on the wiki page, or if there's a bug that makes it unoperable.

zoggins, 2020-12-15:
Is it a official Nano or a clone?  I found out recently that official Nanos have issues.  I am working on hunting down the problem.  If it's not an official one, what board is it exactly?

Sent from my Verizon, Samsung Galaxy smartphone

Ch0wW, 2020-12-15:
It's an Arduino Nano with a USB-C connector. It's seen as an USB-SERIAL CH340.
[image]

zoggins, 2020-12-16:
Do you happen to have a link, so I can grab one?

Ch0wW, 2020-12-16:
Of course! I bought it on Aliexpress: https://fr.aliexpress.com/item/1005001524771874.html?spm=a2g0s.9042311.0.0.27426c37uPmqDY

But it's no different than a Nano clone, just with an USB-C connector.

sk84uhlivin, 2020-12-16:
Please note that most Saturn extension cables are poorly made, and this could be a controller-to-cable connection issue. Wiggle the connection around and see if it makes a difference. Also, it'd be a good idea to use a continuity meter to verify that there are no shorts going to your Dupont cables.

Ch0wW, 2020-12-16:
I'm aware most are poorly made, but all buttons are properly working on the console directly, resoldered all the Digital pins and I checked all the connections with my multimeter to really make sure I didn't do anything wrong.

zoggins, 2020-12-16:
I have never actually tested the Saturn with a Nano, so I am trying to repro now.  Will let you know what I find.

zoggins, 2021-12-01:
Still having problems with this?

Ch0wW, 2021-12-01:
I'm sorry, but I still encounter this issue as of today (tested with RetroSpy 4.4)...

zoggins, 2021-12-01:
I am getting an exact duplicate of your nano to see if I can reproduce the problem.  If I can’t then it is probably your cable, but we will cross that bridge when we get to it. 😊  Sorry for the delay.

zoggins, 2021-12-02:
Also what controller are you using?

Ch0wW, 2021-12-02:
> I am getting an exact duplicate of your nano to see if I can reproduce the problem. If I can’t then it is probably your cable, but we will cross that bridge when we get to it. 😊 Sorry for the delay.

I recently bought several Nanos (with USB-C). The model isn't the exact same one as I thought (which is a shame since I built a 3D enclosure case specifically for it), but I can still try to rebuild one using another spare cable to see if it persists.

> Also what controller are you using?

I have the original (Digital) Sega Saturn controller.  I also used the Japan-exclusive Arcade stick (HSS-0136) for my tests.

zoggins, 2021-12-02:
Waiting on the nano and HSS-0136.

zoggins, 2021-12-19:
I just got the board and it works right out of the gate for me.  Let me double check the website that the pinouts are correct. 
These are the pins you are using:
Saturn | Arduino
1 | Not Connected
2 | 3
3 | 2
4 | 6
5 | 7
6 | 8
7 | 5
8 | 4
9 | GND

</body>
</html>

zoggins, 2021-12-19:
If you are getting nothing I would double check the select pins 5->7 and 6->8 to make sure they are connected.  2->3, 3->2, 7->5 and 8->4 are the data lines if one of them was messed up or they were mixed up you would get strange behavior not none.

zoggins, 2022-01-12:
Closing due to inactivity.  Feel free to repopen when needed.

closed 2022-01-12 by zoggins