retrospy/RetroSpy #171: occasional N64 button glitches

alkamaass, 2021-01-05:
https://youtu.be/XOhhDUVIEfU?t=117

I have all the blink reduction options turned on, and I still get occasional button glitches though, is there a specific part of my input display I should be insulating more or something? Or is this a software issue 
[image]

zoggins, 2021-01-05:
Is it by any chance during memory card accesses?  Or possible Rumble triggers?  There are known glitches around those two things, but it could also be something else.  I will take a look at it.

https://youtu.be/XOhhDUVIEfU?t=117

[image]<https://user-images.githubusercontent.com/77021109/103685400-28334900-4f5b-11eb-9718-73194350e644.png>

zoggins, 2021-12-09:
closing from inactivity.  Feel free to repopen if necessary.

Haplo064, 2022-07-10:
Heya, I come with more information on this.
The majority of games work fine, but some games have repeated read inputs, even when not doing any rumble or memory card reading. This is from the main menu of Jet Force Gemini, for example.

[image]
.

zoggins, 2022-11-24:
After some testing this is a known issue of using the Arduino to spy on the N64.  The arduino is just barely fast enough to handle the datastream from the controller and in some cases it just can't do it right hence the random inputs.  The only 100% way to ensure this not happening is to switch to a Teensy 3.5 for spying instead of Arduino, which is much much faster.  unfortunately right now Teensy's are really hard to find, but its an option when they become available again.

Haplo064, 2022-11-25:
Hi Zoggins,

Is there anything preventing trying it out on a Teensy 4.1? The board is basically the same, just faster. Plus they're in stock compared to the 3.5.

zoggins, 2022-11-25:
I just ordered one.  I know it will require some minor code changes, but I am pretty sure I can get it working.

zoggins, 2022-11-25:
I got it working with a Teensy 4.0.  I have a Teensy 4.1 on the way to test, but it should be fine.  I will do a release in a few days.  I will also likely port Gamecube also, which suffers from the same problems.