retrospy/RetroSpy #172: FYI, found the V1 GameBoyPrinterEmulator.cpp in your codebase is uncredited and not GPL v3.0 compliant and copyrighted to RetroSpy not arduino-gameboy-printer-emulator.

mofosyne, 2021-01-31:
In https://github.com/retrospy/RetroSpy/blob/master/firmware/sketches/GameBoyPrinterEmulator.cpp#L35 found the code I written attributed to Christopher "Zoggins" Mallery.

Requesting that it be noted as a wrapper over https://github.com/mofosyne/arduino-gameboy-printer-emulator and correctly attributed as being under `GNU General Public License v3.0` (If I understand this is typically viral and thus would demand the whole codebase be under v3.0, but a reasonable interpretation would be for this specific offending section be corrected).

And that the authors is myself Brian Khuu as well as contributors being from https://github.com/mofosyne/arduino-gameboy-printer-emulator#contributors--thanks . Please also note this in your licence and any relevant pages so people can discover the main codebase for this project.

https://github.com/mofosyne/arduino-gameboy-printer-emulator

Anyway I do think your project is pretty cool, but just want to make sure that these particulars is fixed to ensure that my work and others who helped with this project is credited correctly such as HerrZatacke, crizzlycruz, and others

zoggins, 2021-01-31:
No offense was intended, will get it fixed.

zoggins, 2021-01-31:
Should be better now.  If there is more you would like me to do just let me know.  I really meant offense, your project is awesome.

closed 2021-01-31 by zoggins

mofosyne, 2021-02-01:
Yeah that's reasonable enough. It's not just a one-way benefit. Stuff like the https://github.com/mofosyne/arduino-gameboy-printer-emulator/blob/master/showcase/showcase.md showcase page also helps people to discover my stuff more via google searches as well as help foster a good community of people over at https://t.me/gameboycamera and other communities as well. Which means more contributions!

So no offense taken and hopefully you can join in on the fun.