retrospy/RetroSpy #174: Specific settings per type in skin.xml

edru, 2021-02-22:
Would like to request a way to load/ignore specific settings in a skin config file.

Reasoning: 
Certain controllers conflict with each other - biggest culprit is that the Nintendo and Microsoft use the same letters, but on the opposite buttons.
Certain controllers are held differently, like when using a SNES controller for NES you wont be pressing B & A, youd press Y & B.
This would allow for a more universal skin setup.

Ideas: 

dpad assignments could be universal since they always match correctly

Create xml tags for NES, PCGeneric, SNES that would only load that specific section when that specific type is selected, ignoring the rest of the config. It would requiring some redundant skin config as you'd have to declare buttons multiple times, but more universal in the end.

40wattrange, 2021-11-09:
I believe this was done, but I couldnt find the documentation that you gave me - if you still have it could you post it in the skins readme, or even here?

zoggins, 2021-11-09:
Should be able to look at the wii-snes skin for how to do it.

closed 2021-11-17 by 40wattrange