retrospy/RetroSpy #177: MiSTer multiple controller support

edru, 2021-03-11:
Similar to how you can select various controllers in PC Generic, I would like to request teh ability to select from various controllers in MiSTer.

zoggins, 2021-03-11:
This is probably possible, let me pull out the MiSTer and look.

Sent from my Verizon, Samsung Galaxy smartphone

zoggins, 2021-12-01:
What exactly was the ask here?  Was this for selecting the controler ID?]

40wattrange, 2021-12-01:
This specific ask was ability to choose which controller to spy, similar to how you can with xinput 0-3 options. Gives you the ability to pick in case you have multiple controllers plugged in.

zoggins, 2021-12-01:
Yes that should be in the latest release

zoggins, 2021-12-02:
This is supported in the newest release.

closed 2021-12-02 by zoggins