retrospy/RetroSpy #178: MiSTer Multiplayer support

edru, 2021-03-11:
In conjunction with supporting multiple controllers plugged in, it would be nice for MiSTer to send off commands if more than one controller was in use.

This way you could spawn multiple clients, have them select different controllers, and still work.

zoggins, 2021-12-01:
I think this might actually work with the controller selector stuff I added.  I need to test this.

40wattrange, 2021-12-01:
I guess this would also play with the other issue I brought up - or may just be one in the same. If you had the ability to pick the controller being spied, then you could launch a 2nd instance and have it choose a different. Feel free to merge this request with the similar ask of controller selecting.

zoggins, 2021-12-04:
This works in the next release.

closed 2021-12-09 by zoggins