retrospy/RetroSpy #179: N64 Firmware - Unsupported games

MrGamy, 2021-03-25:
When running "Pokemon Snap" the Arduino is not able to capture any signals from the controller. In other games like "Super Mario 64" capturing works like a charm.
Are there any troubleshooting steps I could try or is this a general bug in the firmware?

I would be happy to supply any data to help fix the problem. I would need instructions though.

zoggins, 2021-03-25:
Let me see if I can repro, my guess is that the game is doing something funny with the data packet and the Arduino doesn’t recognize it as controller data.  I just encountered this with Luigi’s Mansion the GameCube. Are you using the latest firmware (not that it should matter, just curious)?

MrGamy, 2021-03-25:
I am currently using the retrospy firmware v4.1

zoggins, 2021-03-25:
I should be able to investigate this weekend, maybe tomorrow.  My guess is the packet header from the controller is different for Pokemon Snap (for whatever reason).  I am guessing it’s a relatively simple fix I just need to figure out the new header.

MrGamy, 2021-03-25:
Take your time, no worries :-)
thanks for investigating! awesome project <3

MrGamy, 2021-04-07:
Hey Zoggins, any update on the issue? Can you reproduce it?

zoggins, 2021-04-07:
Sorry for the delay, I should be able to look at it soon. I haven’t forgotten about you, I promise. 😊

MrGamy, 2021-04-07:
No worries! Sorry for poking. Your estimate sounded like a 5min lookup that's why I was following up.
I saw that you were working on retrospy over on discord - so I was wondering if you couldn't reproduce the problem.

zoggins, 2021-04-07:
I can definitely repro.  I think I know what is going on, but I will need a few days to figure it out.

MrGamy, 2021-04-07:
Thanks for letting me know! Otherwise I would have had to look at my arduino setup :-)

zoggins, 2021-04-07:
Well there is a special place in hell for the hack I had to do to fix it, but I think I have it fixed.  I need to clean it up a bit, but I will release tomorrow.

MrGamy, 2021-04-07:
:D I'd be super interested in what the problem was if you want to get into the details. :)
Thanks a lot for looking into this!

zoggins, 2021-04-08:
Here is the update file if you want to beta test: https://github.com/retrospy/RetroSpy/blob/n64/firmware/sketches/N64.cpp

The long story short is the game seems to smash together a GetControllerInfo and a Controller Data Packet.  Usually it goes Header -> GetControllerInfo -> Header -> Controller Data, but this game seems to go Header -> GetControllerInfo -> Controller Data.  There is no second header, so I was completely throwing away the controller data packet.

MrGamy, 2021-04-08:
Still works on Super Mario 64, still doesn't work with Pokemon Snap sadly just tried both

zoggins, 2021-04-08:
Hmm, I probably broke something

zoggins, 2021-04-08:
Sorry there was a bug in my refactor.  If you could try again, just to check.

zoggins, 2021-04-08:
> it works! 😄
> not sure if it has some lag to it - but I will keep an eye on it

It shouldn’t have any more lag than on other games, but let me know if you see anything.

MrGamy, 2021-04-19:
After some additional testing I noticed that it now works with my Gamecube Controller (via raphnet adapter) but not with my N64 controller (w/o Adapter)
I don't really know why.

zoggins, 2021-04-19:
That is weird.  I only tested with the N64 controller directly to the N64.  I will double check my testing.  Trying to think why it would work with the adapter and not the controller directly.

MrGamy, 2021-04-19:
I agree. It is very weird that it works with the GameCube Controller and 
not the original N64 controller. I was suspecting cable-lenght, but since I 
was able to play the game with the N64 controller I scraped that thought again.

zoggins, 2021-04-19:
Do you have anything plugged into the controller (memory pak, rumble pak)?

MrGamy, 2021-04-19:
Nope, nothing plugged in.
Same pattern as in the beginning though - Controller works in Super Mario 64.

MrGamy, 2021-04-19:
> US N64 or Japanese or European? RAM extension? I am just trying to
> determine the difference between my setup and yours.

Modded japanese n64 + RAM extension (oem)

zoggins, 2021-04-19:
That’s what I have too.  I just retested and Pokemon Snap is definitely still working on my end.  I wanted to make sure I didn’t break something since I last tested.  Which version off of GitHub are you using?

MrGamy, 2021-04-19:
I'm currently running the 4.2 pre-release from a week ago. But will try 4.2.3 as soon as I can

closed 2021-09-20 by MrGamy