retrospy/RetroSpy #180: USB - SNES Controller on PC unsupported?

The3X-cmd, 2021-04-07:
Hey guys.
I tried using a snes controller (8BitDo or iNNEXT) on PC.
First weird thing is, I cannot find any style preset for those.

Second is, once I tried to alter the "PC - Generic Pad", I discovered that the "UP, DOWN, LEFT, RIGHT" Buttons are not entitled like that, but are x and y axes.
And therefore only a stick display makes sense.
However this is very weird.
When I try it with "button x+", then while pressing right or left, the y axes are pressed jsut a little. and while pressing top or buttom the x axes is just pressed a little. So it does not work with "x+,x-,y+,y-".
Did I somehow install the controller wrong? Is there a better way?
Howto use this :(
I don't know where to ask, I tried it for hours....

Additionally: 
Whenever I now (since newest update) start the program, it calls alot of errors at start, for example for using "precise" identifier.
I uninstalled and reinstalled many times, still those errors persist without any change I made myself.

zoggins, 2021-04-07:
Specifically what controllers have you tried?  The PC controllers can have weird mappings sometimes.

The3X-cmd, 2021-04-07:
Like the examples I wrote...
Now I'm trying 8BitDo SN30.
I have problems using
```
    <button name="y-" image="circle.png" x="46" y="41" width="16" height="16"/>
    <button name="y+" image="circle.png" x="46" y="69" width="16" height="16"/>
    <button name="x-" image="circle.png" x="32" y="55" width="16" height="16"/>
    <button name="x+" image="circle.png" x="60" y="55" width="16" height="16"/>
```
Because, as I see in the "Generic Skin Helper" whenever I press "right" or "left" for the "x" axes, my y axes changes just a little bit to the minus area.
Same happens for the x axes when I press "up" or "down".

I wonder if I can someone make a minimum amount of pressure that is needed. That would totally solve the problem.
Or if I somehow can change the mapping (button names in retrospy) so these are no longer axes but buttons.

zoggins, 2021-04-07:
I have a SN30 Pro (wired) and when I plug it in detects as an Xbox 360 controller and it just works.  What is Windows detecting the controller as?

zoggins, 2021-04-07:
Also is this the controller?

https://www.amazon.com/Wireless-Controller-Classic-nintendo_super_NES-super-nintendo/dp/B074HBNNH6/ref=sr_1_6?dchild=1&keywords=8bitdo+SN30&qid=1617767222&sr=8-6

The3X-cmd, 2021-04-07:
Just another colour, but it is the controller.
The controller works fine for the game, I believe it is detected as XBox controller.
However I want to show the inputs on the controller.
I do not want to use 
`    <stick xname="x" yname="y" image="circle.png"
           x="53" y="31" width="34" height="35" xrange="9" yrange="9" yreverse="true" />`
Because this does mean, that there is always like a dot in the middle or sth.
It is just not possible to show the inputs the way I like it.
The way the normal snes controller would do.
The normal one would show a dot on the key pressed. i.e. on the up and left key there would be a dot if I press up left.

But if I use the "stick" method, then the dot would be between left and up.
And there will be a dot in the middle of the dpad always.

zoggins, 2021-04-07:
With the sn30 pro Up/Down/Left/Right don’t affect the analog sticks and show correctly as they would on an SNES.  So there must be something different, let me order a controller and I will figure it out.

The3X-cmd, 2021-04-07:
Okay the main problem actually is, that when I press UP or DOWN, that the "x-" is triggered a little bit.
And when I press LEFT or RIGHT the "y-" is triggered a little bit.
I can see it on the display helper. It is just pressed a very little too.
My game does not react to that little, but apperently retrospy does.

The exact product is this:
https://www.amazon.de/gp/product/B07GK8LJBJ

zoggins, 2021-04-07:
I will figure it out 😊 Thanks for finding the problem.

https://www.amazon.de/gp/product/B07GK8LJBJ

zoggins, 2021-04-09:
SN30 on PC in D-Input Mode (the default or hold B+start for more than 3 seconds) will be natively supported in the next release which should be out next week.

zoggins, 2021-12-01:
This is fixed.

closed 2021-12-01 by zoggins