retrospy/RetroSpy #184: Seeing analog input (Stick or analog trigger) in digit output.

Robert-Ordis, 2021-06-13:
Is there any way to see the input of analog stick / analog trigger in DIGIT OUTPUT ?

I need to see the input value in digit for setup the neutral position of the manual super swim in Legend of Zelda: The Wind Waker.

zoggins, 2021-06-13:
So you want the analog sticks to show a button press on the input display?

zoggins, 2021-06-13:
Oh I think I miss understood.  You want to see the actual number of the analog stick/trigger?

Robert-Ordis, 2021-06-13:
yes

zoggins, 2021-06-13:
I can probably put something together.  Let me see what I can do.

Robert-Ordis, 2021-06-13:
Please.
I'm looking forward.

zoggins, 2021-07-16:
I am starting work on this now, sorry for the wait.

zoggins, 2021-07-23:
in the next release you will be able to do this:

<analogtext name="lstick_x" x="61" y="2" font="Arial" size="20" color="#FFFF00" range="100" />

that will represent the lstick (or whatever analog input) as a value -100 to 100 (in this case).  Obviously a trigger would be 0 to 100.  The scale can be changed with the range value.

zoggins, 2021-08-03:
as shown above in 4.3

closed 2021-08-03 by zoggins

Robert-Ordis, 2021-08-03:
Thanks a lot ! it works fine !