retrospy/RetroSpy #185: [Documentation Request] How to reverse engineer an unsupported controller?

wertercatt, 2021-07-08:
I think documentation on how to figure out the signals of a currently unsupported controller would be very useful, as it would allow people who own more obscure and unsupported controllers to contribute to the project with their findings.

zoggins, 2021-07-08:
That is a good idea.  I actually have two vtech kids “computers”  that I was going to reverse engineer, maybe I should document what I do. 😊  Is there a controller you had in mind?

wertercatt, 2021-07-08:
I'm planning on eventually getting myself a Nuon, but I was also interested in reversing the signals of the currently unsupported Xbox controllers, such as the remote.

wertercatt, 2021-07-08:
I'm also wondering if I could hook the Wii's Bluetooth adapter up to RetroSpy, as it communicates over a standard USB interface. This would potentially allow Wiimote input display, not just the extension controllers.

closed 2021-12-01 by zoggins