retrospy/RetroSpy #189: WMI Provider Host causes high CPU usage (Windows 10)

maggot9779, 2021-07-19:
After running the software, WMI Provider Host jumps to around 20-30% CPU usage in the Task Manager. I know it is RetroSpy causing the issue because if I close the software, WMI Provider Host goes back to normal low levels. 

It doesn't matter if the software is displaying the skin or not, the CPU usage is still high.

I've tried version 4.1 and 4.2.3, there's no difference. 

This does not happen with NintendoSpy.

zoggins, 2021-07-19:
I see it jump to 10% on my box, but I will take a look at what is causing it.  Thanks!

zoggins, 2021-07-19:
What is interesting is that it happens at the setup window before any spying starts.

zoggins, 2021-07-19:
I bet I know what is causing it, but I need to verify.  It’s the code that auto detects the Arduino/Teensy in the setup window.  The odd thing is that is seems to continue running once the setup window has closed.

zoggins, 2021-07-19:
Hmmm its not that either.  I will continue to investigate.

zoggins, 2021-07-19:
Found it.  The timer that polls the serial ports now stops once the viewer starts.  This will go out in the next release which will be soon.

maggot9779, 2021-07-19:
Nice. Thanks for looking into it so quickly.

closed 2021-07-19 by maggot9779