retrospy/RetroSpy #190: Question: Is it possible to define a shortcut (or flags) to directly open the Viewer?

ajshepley, 2021-07-24:
Hello,

I usually start RetroSpy with the intent of launching the same viewer for the same controller + skin + COM port as my last session.

I'm wondering if there's a way to launch directly into the viewer with the previous settings, e.g. from a windows shortcut or CLI flags.

I tried looking through the closed/open Issues and previous PRs and the viewer through Process Explorer but didn't see anything obvious.

If not, then I suppose this is also a bit of a feature request, but I haven't checked through the code so I don't know how (in)feasible this would be.

zoggins, 2021-07-24:
There is no way to do it now, but I will look into adding it.  Probably shouldn’t be too bad.  Would like a command line flag to just do everything the same as last time be sufficient?

zoggins, 2021-07-25:
4.3 will have a commandline flag -skipSetup where it will go straight to the view window using the previous saved defaults.

ajshepley, 2021-07-26:
I think a command line flag would work well. That should work both when calling the executable on a terminal, and from a Windows shortcut/start menu icon (users can set params in Target in right click -> properties on a shortcut). Thank you very much!

zoggins, 2021-08-02:
Newest version has -skipSetup command line flag

closed 2021-08-02 by zoggins