retrospy/RetroSpy #192: remove broken skins

MrGamy, 2021-07-29:
Some skins seem to have stopped working for a while now and should be repaired or removed since they are causing an error message when starting retrospy:

    generic-hitbox-brookeps3ps4tomdsc-ds4-md6b
    generic-hitbox-brooksc-ds4-md6
    generic-hitbox-ds4
    generic-hitbox-xb360
    generic-pattonboxdigital-brookeps3ps4tomdsc-ds4-md6b
    generic-pattonboxdigital-brooksc-ds4-md6
    generic-pattonboxdigital-ds4
[image]

zoggins, 2021-07-29:
In the Help -> About screen what version does it say.  The precision keyword is only supported from 4.1 and up.  Not sure how you got the new skins, but I am betting your .exe is (for some reason) below 4.1.

[brokenskins]<https://user-images.githubusercontent.com/8639512/127480649-5e3e5053-1807-44b1-a21a-e3f9741baa15.png>

MrGamy, 2021-07-29:
I am running Version 4.2.3.0
Build Timestamp is 30/07/2021 00:03:12
[image]

zoggins, 2021-07-29:
Hmmm, that is unexpected.  I will see what I can find, but I can’t repro the error.

[about]<https://user-images.githubusercontent.com/8639512/127571809-95f7378d-7c90-4661-812a-6deb3db55fd3.png>

zoggins, 2021-07-29:
Did you delete the skins?  If you didn’t delete them could I get a copy of one of them?

MrGamy, 2021-07-30:
here you go - extracted one of the skins but added my whole retrospy folder for good measure.
https://drive.google.com/file/d/1v7cq3xjwguG0pBGLidSd0M-r4AY1Hcf3/view?usp=sharing

MrGamy, 2021-08-02:
I included my whole retrospy folder in this archive as well. Hope it helps :-)
https://drive.google.com/file/d/1v7cq3xjwguG0pBGLidSd0M-r4AY1Hcf3/view?usp=sharing

MrGamy, 2021-08-02:
Quick update - I just downloaded the current retrospy-release.zip and opened that fresh install: same error message.
precision seems not to work

zoggins, 2021-08-02:
Well to add to the confusion I just downloaded your zipped up RetroSpy directory and it starts up fine with the skin in the skin folder.

MrGamy, 2021-08-02:
haha, awesome! 
That doesn't really help with finding out whats wrong though. D:

zoggins, 2021-12-01:
still having the problem?

zoggins, 2021-12-09:
closing from inactivity.  Feel free to repopen if necessary.

MrGamy, 2021-12-10:
sorry for not responding earlier - yes, this issue is still persistent, I have not checked the nightly build though.
Might there be any dependency I need to have installed on windows?

zoggins, 2021-12-10:
Another person is having this issue.

zoggins, 2021-12-10:
What language is your Windows set to?  Is it English or something else?

MrGamy, 2021-12-10:
My Windows is set to English currently. Secondary Language would be German

zoggins, 2021-12-10:
When you get a chance could you try this build: https://github.com/retrospy/RetroSpy/releases/tag/diag1

zoggins, 2021-12-12:
Finally solved this.  It had to do with locale settings expecting a ',' instead of '.' in float numbers.

closed 2021-12-12 by zoggins