retrospy/RetroSpy #193: Nintendo NES option has latency on displaying inputs

Kosmic12, 2021-08-18:
When using the "Nintendo NES" option to display inputs, there is a significant amount of latency between when the input happens and when it is displayed on RetroSpy. Images below show the exact amount. This delay doesn't happen with any of the other options I've tried (SNES, N64, GCN), even though all my devices are using the same firmware. Several other people have shared this experience with me as well.

Here is the moment the jump input happens in game:
[image]

Here is the first frame the input is displayed on retrospy (about 15 frames of delay, or on average .260 seconds):
[image]

zoggins, 2021-08-18:
What Arduino board are you using?  There has been some reports of latency in NES and I haven’t been able to track down what is common between them.

[image]<https://user-images.githubusercontent.com/41668414/129963806-1661065d-34fc-4c13-a620-63f5c952a8d3.png>

[image]<https://user-images.githubusercontent.com/41668414/129964068-d586128d-f9a2-4db5-bbef-032bc0e3edad.png>

zoggins, 2021-08-18:
Also what NES/Famicom type?  Flash cart or actual cartridge?

[image]<https://user-images.githubusercontent.com/41668414/129963806-1661065d-34fc-4c13-a620-63f5c952a8d3.png>

[image]<https://user-images.githubusercontent.com/41668414/129964068-d586128d-f9a2-4db5-bbef-032bc0e3edad.png>

zoggins, 2021-08-18:
Also, is it all games or only certain ones?  Is it all the time or some of the time?

[image]<https://user-images.githubusercontent.com/41668414/129963806-1661065d-34fc-4c13-a620-63f5c952a8d3.png>

[image]<https://user-images.githubusercontent.com/41668414/129964068-d586128d-f9a2-4db5-bbef-032bc0e3edad.png>

Kosmic12, 2021-08-18:
Great question about different games, I had only studied the frames closely with SMB1 so I didn't realize others are actually different. I tried Ninja Gaiden and it appears to be 5 frames delay. Contra was also a consistent 5 frame delay. It seems SMB1 is probably the worst offender at ~15. 

As far as consoles and flash cartridges go, I've experienced the same delay between toploader NES, and AV Famicom. The delay is also the same between original cartridge, PowerPak, and Everdrive Pro.

The arduino board is "ELEGOO Nano Board CH340/ATmega+328P" and with CH340 USB chipset.

Thank you!

zoggins, 2021-08-18:
Something to try as I have seen this issue with some Nanos.  Did you program it yourself?  If so you can try to uncomment out #define LAG_FIX in the firmare.ino file (its towards the top).  Then use the “Options -> Need Lag Fix” in the UI.  That might help the problem.

Kosmic12, 2021-08-18:
aha, that did the trick! 1-3 frames delay in SMB1 now. I'll take it! Thank you!

zoggins, 2021-08-18:
Yeah its something about some Nanos that cause an issue.  I am currently brainstorming ways to make the detection more automatic.  Glad I could help.

closed 2021-12-01 by zoggins