retrospy/RetroSpy #194: Input display fails when connected to MiSTer SNAC adapter

xexzy, 2021-08-19:
When attempting to use an input display and a SNAC adapter with the NES core, improper inputs are read. With one controller, all buttons were read as being pressed, another controller resulted in various buttons other than what was pressed, another controller resulted in no inputs being read, and a dog bone (nes-039). If an input display is not in place, the SNAC adapter works. The input display also functions properly when plugged into my NES or into a USB adapter then plugged into the MiSTer as a USB controller.

It was suggested to try flashing the SNES firmware but use a NES controller and this also failed.

zoggins, 2021-08-19:
So its actually scrambling inputs or just scrambling the display of inputs?  I will get out my SNAC adapter and see if I can repro.  What Arduino are you using?  What version of the MiSTer I/O board?

zoggins, 2021-08-19:
Well the bad news is I can’t repro on my end.  Does your cable connect to Arduino GND?  Does behavior change in different games?  I am sure we can get to the bottom of this.  Just trying to figure out what is different between your setup and mine.

xexzy, 2021-08-19:
the only controller I have been able to get to function properly with both the mister and the retrospy is the stock dogbone. I've noticed that different chips within oem controllers have different results. Here's a short video where the retrospy is showing the correct inputs but the mister is reading incorrect inputs https://youtu.be/1dA46yPqrm4 my mister version is 210803 os v210711 and my retrospy should have the latest firmware on it. I'm sorry to say I don't know for sure if the ground is connected to the arduino ground. I didn't put these together, but both of mine do not look like they have a ground wired to the arduino ground. (thank you for your prompt response too)

zoggins, 2021-08-19:
All I have to test with is stock OEM rectangle controllers?  Do you have a picture of a misbehaving controller?  I might be able to get one to test.

xexzy, 2021-08-19:
The one in that video is an oem brick that has an aftermarket IC to make it function again (chip was purchased at https://www.amazon.com/gp/product/B082X7BWNR/ref=ppx_yo_dt_b_asin_title_o01_s00?ie=UTF8&psc=1). The other controller that behaves the same way has the same replacement chip. Another controller that shows all buttons being pressed (on both the mister and the input display) has a MC1421BCP IC. I believe this is an OEM board. And the dogbone has a 4021BF IC. If you would like picture of the inside I will try to make that happen.

40wattrange, 2021-11-09:
Just to leave a note here, this could be at fault of the SNAC adapter. I know there was a version that was out there for a while that had improper chips on it, and OEM controllers didnt work well with it at all and was later revised.

It could be possible that this was the case in this situation if you cant reproduce it.

I know the SNAC version I have does not work with RetroBit Genesis controllers at all.

zoggins, 2021-12-01:
Is this still aproblem

zoggins, 2021-12-09:
closing from inactivity.  Feel free to repopen if necessary.