retrospy/RetroSpy #199: PS1 controller no longuer outputs

Avasam, 2021-10-02:
It's been a long this since I last used my PS2 with input display. So I'm not sure which version exactly broke support.  
My PS1 analog controller on a PS2 has stopped showing any inputs on RetroSpy.  
In the version that used to work, the skin was called "Playstation Analog in Red Mode"  
I'm currently using v4.3.1  

https://user-images.githubusercontent.com/1350584/135732411-7b305481-0511-4682-997d-4a613b03c7aa.mp4

zoggins, 2021-10-02:
I will take a look.  In 4.4 I changed some PlayStation stuff, but I don’t think I messed with the PS code in 4.3.1.  And by analog controller you mean the actual analog controller do you mean a dualshock 2?

https://user-images.githubusercontent.com/1350584/135732411-7b305481-0511-4682-997d-4a613b03c7aa.mp4

Avasam, 2021-10-02:
I use a DualShock 1 with a Playstation 2 Slim

zoggins, 2021-12-01:
is this still an issue?

closed 2021-12-01 by zoggins

Avasam, 2021-12-06:
> is this still an issue?

Yes, I bissected and 2.6 is the version that broke for me.
2.5 works fine.

zoggins, 2021-12-06:
Have you flashed your Arduino with the newest firmware?

Avasam, 2021-12-06:
I've just tried following the guide at https://retro-spy.com/wiki/playstation-1-2-on-arduino-getting-started/
But I'm always getting the following error when trying to upload firmware. I did unplug everything from the Input Viewer, plugged the USB back in and validated the COM PORT
```
An error occurred while uploading the sketch
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 1 of 10: not in sync: resp=0xb6
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 2 of 10: not in sync: resp=0xb6
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 3 of 10: not in sync: resp=0xb6
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 4 of 10: not in sync: resp=0xb6
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 5 of 10: not in sync: resp=0xb6
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 6 of 10: not in sync: resp=0xb6
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 7 of 10: not in sync: resp=0xb6
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 8 of 10: not in sync: resp=0xb6
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 9 of 10: not in sync: resp=0xb6
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 10 of 10: not in sync: resp=0xb6
```
[image]

zoggins, 2021-12-06:
I have never seen that before.  Something to try and right after you hit upload in the GUI immediately hit the reset button on the Arduino.

Avasam, 2021-12-06:
And here I was hoping I wouldn't have to open my tape and cardboard "case" XD
It also turns out it's a pre-2018 Nano that I have. Either selecting the right board + processor AND/OR hitting the reset button made it work.

Both DS1 and DS2 are detected on the latest version now. Huge thanks for the support!

zoggins, 2021-12-06:
Awesome!! Glad I could help!