retrospy/RetroSpy #202: SG-1000/SC-3000 controller support

Shentokk, 2021-11-07:
It seems that the SMS protocol doesn't work for SG-1000 controllers though the pinout seems the same from research I've done.

zoggins, 2021-11-07:
Really?  The SMS protocol is super simple, so this surprises me.  I will see what I can find.

Shentokk, 2021-11-07:
I do wonder if it's something on my end then. The TX light on the Uno is permanently lit instead of blinking like with other profiles. I'm using the newest firmware and the other modes don't have this issue.

zoggins, 2021-11-07:
TX light on implies the Arduino is sending data to the PC, but that is normal for the DB9 protocol. Their isn’t really even a protocol for SMS, each wire represents a button/direction.  So I am wondering why the Arduino isn’t getting the signals.  When you load the viewer there is just no activity at all?

Shentokk, 2021-11-07:
That's correct. There's absolutely no activity when connected to the SC-3000.

zoggins, 2021-11-07:
What controller are you using?

Shentokk, 2021-11-07:
I am using a Sega SJ-150 controller. I've also tried my SJ-152 Mark III controller.

https://segaretro.org/Joypad_(SJ-150)

zoggins, 2021-11-07:
I should ask, what console/computer is the problem?

Shentokk, 2021-11-07:
I am using a Sega SC-3000 computer.

zoggins, 2021-12-01:
So the controller works fine an on SMS just not the SC-3000?  Or does the controller not work on anything?

Shentokk, 2021-12-03:
The controller works fine when used on a SMS.

zoggins, 2021-12-08:
Wait what viewer are you using in the UI?  If you are using MODE_SMS_ON_GENESIS you need to select Genesis the UI even if you are on a Master System

Shentokk, 2021-12-12:
`<?xml version="1.0" encoding="UTF-8"?>
<skin name="SMS" author="Shentok" type="genesis">

    <background name="SMS" image="gamegear.png" />

    <button name="1" image="button.png" x="311" y="192" width="79" height="79" />
        <button name="red" image="button.png" x="311" y="192" width="79" height="79" />
            <button name="a" image="button.png" x="311" y="192" width="79" height="79" />
    <button name="2" image="button.png" x="408" y="117" width="79" height="79" />
        <button name="blue" image="button.png" x="408" y="117" width="79" height="79" />
            <button name="start" image="button.png" x="408" y="117" width="79" height="79" />

    <button name="up" image="up.png" x="71" y="68" width="56" height="78" />
    <button name="down" image="down.png" x="71" y="151" width="56" height="78" />
    <button name="left" image="left.png" x="20" y="122" width="76" height="53" />
    <button name="right" image="right.png" x="102" y="122" width="79" height="53" />
	
</skin>
`

I am using it in the genesis section.

zoggins, 2021-12-12:
So my setup is:
SC-3000 playing Star Jacker
Arduino UNO programmed with MODE_SMS_ON_GENESIS
Sega Genesis -> SMS Controller skin
SJ-150 controller
Pinout:
Genesis Pin 1 -> Uno pin D2
Genesis Pin 2 -> Uno pin D3
Genesis Pin 3 -> Uno pin D4
Genesis Pin 4 -> Uno pin D5
Genesis Pin 5 -> Not Connected
Genesis Pin 6 -> Uno pin D6
Genesis Pin 7 -> Uno pin D8
Genesis Pin 8 -> Uno GND
Genesis Pin 9 -> Uno pin D7

and it works, so something strange is going on.  Is this your setup also?

Shentokk, 2021-12-12:
So it turns out that it was because I didn't have ground connected. It had gotten loose on the breakout board and it looked like it was still attached, but it wasn't on closer inspection. Fixing that did the trick. Sorry for all the trouble on something so simple.

zoggins, 2021-12-12:
No worres, glad its working now

closed 2021-12-12 by zoggins