retrospy/RetroSpy #204: Ability to reassign Genesis PINs in config_arduino.h

40wattrange, 2021-11-09:
I see that most, if not all, controllers have the ability to be repinned via the config file. I only see the following options for Genesis:

#define GENESIS_TL         DIGITAL_PIN_06
#define GENESIS_TR         DIGITAL_PIN_07
#define GENESIS_TH         DIGITAL_PIN_08

Compared to the number of wires you need to get the genesis working I'd assume there is more. Can we get the documentation on that as well as what each means?  Personally not familiar with what TL TR TH is in relation to function.

If this is not in the build currently, could be possible get that ability like the other controllers?

40wattrange, 2021-11-17:
In addition to the Genesis pins - I would like to add the Game Boy link cable.

40wattrange, 2021-12-01:
The priority systems for myself would be Genesis as top and Game Boy Printer 2nd.

These 2 things would bring the ability for me to make the SNAC sniffer 100% compatible w/ RetroSpy.

zoggins, 2021-12-01:
Genesis is complicated I will have to look at it a little bit more.

The Gameboy printer pins can be changed in gbp_emulator_v2.cpp

#define GBP_VCC_PIN               // Pin 1            : 5.0V (Unused)
#define GBP_SO_PIN        4       // Pin 2            : Serial OUTPUT
#define GBP_SI_PIN        3       // Pin 3            : Serial INPUT
#define GBP_SD_PIN                // Pin 4            : Serial Data  (Unused)
#define GBP_SC_PIN        2       // Pin 5            : Serial Clock (Interrupt)
#define GBP_GND_PIN               // Pin 6            : GND (Attach to GND Pin)
#define LED_STATUS_PIN   13       // Internal LED blink on packet reception

I haven’t tested changing them, so your milage may vary.  I will look at moving this to the config*.h files.