retrospy/RetroSpy #205: RetroSpy crashes immediately

CommonPuffin, 2021-11-14:
Hello. I'm trying to use RetroSpy to display my inputs on streaming.
It worked (likely) fine on the Beaglebone side, but I had a problem on the PC side.
I downloaded the latest version of RetroSpy-Setup.exe and completed the installation without error.
But when I try to run RetroSpy, I get a pop-up message that says "RetroSpy has stopped working" "A problem caused the program to stop working correctly. Please close the program.".

What I have tried is to

- run RetroSpy as an admininstrator
- restart PC
- reinstall RetroSpy
- install older versions
- check for windows updates
- plug / unplug my mouse, keyboard, Ethernet, USBs etc

and unfortunately, none of them worked. Am I doing something wrong? 

---

Edition    Windows 10 Pro
Version    21H1
OS Build    19043.1348
Experience    Windows Feature Experience Pack 120.2212.3920.0

[crash_event.txt (Roughly translated because it was not in English)](https://github.com/retrospy/RetroSpy/files/7532952/crash_event.txt)

zoggins, 2021-11-14:
I will take a look

zoggins, 2021-11-14:
Do you have ActiveX installed?  Should be, but it looks like the ActiveX stuff that makes keyboard/mouse spying possible is crashing.  I will look into what would make that crash. In the mean time I will put together a build that skips those readers, assuming you don’t need it.

CommonPuffin, 2021-11-14:
Thank you for the prompt response.
Strange, I don't remember changing any of my ActiveX settings...
Is there anything I can do on my side?

40wattrange, 2021-11-17:
Not sure if you have the same issue as me, but mine was crashing. I found that if you have a skin that is configured to use multiple types, and one of the types is generic, it will crash the program immediately.

zoggins, 2021-12-01:
Try this release.  It will bring up an error message instead of just crashing.  Let me know what the message is.

https://github.com/retrospy/RetroSpy/releases/tag/diag1

CommonPuffin, 2021-12-04:
Hi, sorry for the late reply.
I just installed and ran the customized retrospy, and oddly enough, the program launched without any problems.
I am a bit confused because I have turned off automatic Windows updates a few years ago, I haven't installed or uninstalled any other software, and I haven't even touched my Windows in the past few days.
I can't vouch for the correctness of this, but the reason that comes to mind as to why the problem has been resolved is the automatic update of the security software database. As I understand it, this is the only factor on my PC that can change over time.
If it was really caused by the security software, the function to block keyloggers may have prevented the program from starting.
For people who have the same problem as me, changing the settings of the security software to trust the program, or temporarily turning off the protection may help.
Thank you for your concern.

closed 2021-12-06 by zoggins