retrospy/RetroSpy #206: MODE_DETECT additions

40wattrange, 2021-11-15:
So a handful of requests to expand support of just one arduino device:

1) The Arduino Nano has analog pins A0-A7 - would it be possible to get the additional analog pins supported rather than limiting it to the 5 of the full?

2) To go one step further to add another "slot" - can we get a mode_detect value of null? Essentially if no analog pins are grounded, boot a specific config. This could act as a "default" state, while allowing another pin being assigned to another system.

3) This is just bonus - if you were to make a "default" state within MODE_DETECT as mentioned in #2 - you could make it so there is no difference between mode_detect and normal modes - mode_dectect could always be on - and then just declare the null state only if you want to do a single system flash.

zoggins, 2021-11-15:
Yeah I can add the extra pins.  There is already a default state.  In the default MODE_DETECT code the else condition (i.e. NES) is chosen if no analog pin is grounded.  and I can probably make 3 work.

40wattrange, 2021-11-15:
oh dang, ok. I need to dive deeper on the mode_detect stuff. It seems to have changed since i last used it.

zoggins, 2021-11-15:
Also while not convenient when using a jumper wire you could theoretically use short multiple analog inputs and treat it as a binary number further expanding the number of consoles you could load at once.

40wattrange, 2021-11-15:
That was something I was going to bring up further, if need be :)

At this time having 8 slots, and another "else" slot is more than enough for me and what im currently doing :)

closed 2021-12-01 by zoggins