retrospy/RetroSpy #209: MODE_DETECT broken?

40wattrange, 2021-11-27:
Downloaded new firwmare
Uncommented the MODE_DETECT option
Verified no other consoles are uncommented

Flash as is and get the failure:

E:\Downloads\RetroSpy-release (1)\firmware\firmware.ino: In function 'void setup()':
firmware:285:16: error: expected unqualified-id before '.' token
         SNESSpy.setup();
                ^
firmware:290:15: error: expected unqualified-id before '.' token
         N64Spy.setup();
               ^
firmware:295:14: error: expected unqualified-id before '.' token
         GCSpy.setup();
              ^
firmware:314:15: error: expected unqualified-id before '.' token
         NESSpy.setup();
               ^
E:\Downloads\RetroSpy-release (1)\firmware\firmware.ino: In function 'void loop()':
firmware:423:16: error: expected unqualified-id before '.' token
         SNESSpy.loop();
                ^
firmware:425:15: error: expected unqualified-id before '.' token
         N64Spy.loop();
               ^
firmware:427:14: error: expected unqualified-id before '.' token
         GCSpy.loop();
              ^
firmware:437:15: error: expected unqualified-id before '.' token
         NESSpy.loop();
               ^
exit status 1
expected unqualified-id before '.' token

Am I doing it wrong?

zoggins, 2021-11-28:
You have to enable at least one other firmware.  And the default code requires the 4 Nintendo consoles (NES, SNES, GC and N64) to be uncommented.

zoggins, 2021-11-29:
[firmware.zip](https://github.com/retrospy/RetroSpy/files/7620627/firmware.zip)
Here is a working example of mode detect for the 4 Nintendo systems.

zoggins, 2021-12-01:
Is this example what you needed. I need to update the documentation

40wattrange, 2021-12-01:
Does this ignore the 4 modes uncommented up top, or are those uncommented because they are the modes being used?

I plan on having:

NES
SNES
SMS
GENESIS
SATURN
TG16
PSX
Game Boy Printer

So just wanna know if i had to uncomment those as well?

zoggins, 2021-12-01:
Then you uncomment each platform’s #define AND the MODE_DETECT #define.

40wattrange, 2021-12-02:
Per the help provided here/discord I believe the question has been answered. Will reach out if more help is needed.

closed 2021-12-02 by 40wattrange