retrospy/RetroSpy #212: Gamecube/Wii: Luigi Mansion (JP) prevents right should analog from detecting

40wattrange, 2021-12-02:
I find that when loading Luigi Mansion (JP) via optical disc on Gamecube or Wii, that the right trigger stops reporting analog signal, only a full press. The left trigger still reports fine. 

Oddly enouogh, this does not happen when loading the US or JP game via ISO on soft modded Wii.

Here is a clip where you can see while the Wii is in the system menu, both triggers are reporting, but as soon as I load the disc, the right trigger drops. Again, it will still report the full click.

https://clips.twitch.tv/CuteCallousCasetteFailFish-mJfJoL5PoSITPAZG

zoggins, 2021-12-02:
That is really strange.  I can burn a JP disc and check it out.

40wattrange, 2021-12-08:
Just did some further testing. Going back to the oldest build on page 1, it does not detect inputs at all. Looking through the changelogs I noticed 4.2.3 specifically mentions addressing Luigi's Mansion inputs. Unfortunately I can't get that specific build to compile and flash to the arduino.

Though some further information on testing with the most recent build, I've noticed that if i do push the R button - the L button reports the analog movement ever so slightly.

In this video you can see the L buttons analog reporting ever so slightly then the R button will click in. So it seems it is sorta being detected and reported in some way.

https://user-images.githubusercontent.com/1192974/145298420-5700bcc0-e183-4575-81c3-1e6bec2c6cce.mp4

.

zoggins, 2021-12-08:
This is using a nano or uno and only from disc?  It doesn’t manifest loading on the USB drive?

40wattrange, 2021-12-08:
My specific case is using a modern Nano that uses new bootloader, with an authentic JP Disc via Wii Menu or Gamecube

Loading the game on a Wii via Nintendont does not generate the issue be it from disc or USB.

Seems to only happen via disc via default load on either system. It is only when you introduce a loader tool on Wii that it goes away.

zoggins, 2021-12-08:
Well the good news is I can repro with my GCLoader and US iso.  So I will figure it out.

zoggins, 2021-12-09:
Found the problem.  Apparently some games (Luigi’s Mansion and Wario World as examples) don’t use 8 bits for each trigger.  They only use 4 and they pack both triggers into the byte normally used by the left trigger.  This will require a firmware update and the newest UI.  I have checked in the code and it will be in the next release and tonight’s nightly build

40wattrange, 2021-12-09:
Confirm the newest nightly resolved the issue.

closed 2021-12-09 by 40wattrange